dotintent/react-native-ble-plx #274: SwiftSupport folder is empty

sookool99, 2018-05-14:
I am trying to upload a project to IOS and is getting rejected because it says

> Invalid Swift Support - The SwiftSupport folder is empty. Rebuild your app using the current public (GM) version of Xcode and resubmit it.

This is the only library I am using that uses Swift and I am not sure how to resolve this.

darkrift, 2018-06-18:
I am in the exact same situation. I actually released a version for iOS last week before I updated to the latest Xcode. I updated Xcode wo include the latest 11.4 ios support and I can't submit to itunes connect ever since. Any idea how to fix this ?

darkrift, 2018-06-18:
I got it back on track after I updated to ble-plx to 0.9.1.

@sookool99 Can you validate that when you run npm install on the command line, it doesn't have an error. It turns out when I updated Xcode to 9.4, Swift library also got updated to 4.1 and the version I was using of plx ble (0.8.0) generated an error because of mismatch Swift version (4.0.2).

Once I solved that compiling error by updating to ble-plx 0.9.1, the app seemed to have been bundled properly as it now passed the itunes connect processing validation.

closed 2019-03-23 by stale[bot]