dotintent/react-native-ble-plx #277: writeCharacteristicWithoutResponse fails occasionally on android, always fails on iphone

pstanton, 2018-05-22:
I'm not sure where the problem lies, but figured it better to post this in-case someone else finds useful.

I've written an implementation which uses this plugin to transmit messages of varying sizes over BLE between handsets (Android and now iPhone) and a bluez/dbus linux server.

Sending larger messages requires that messages are "chunked" into the appropriate size based on MTU.

I noticed that when sending these chunked messages on android using "writeCharacteristicWithoutResponse" the message would occasionally not complete. The first few packets would make it through, but although all were apparently sent from the app, the back end would not receive the rest. Changing this to "writeCharacteristicWithResponse" immediately and consistently resolved the issue.

I've recently worked on the implementation for iPhone, and found that all messages sent using "writeCharacteristicWithoutResponse" would not be received by the back end. Again, changing to "writeCharacteristicWithResponse" fixed the issue immediately.

I'm not sure where the fault is, if it is the implementation of this project, the phone, the BLE chip, dbus or bluez. Feel free to close this issue.

lim6112j, 2019-02-06:
I am making ble device that is communicating images with phone. I have the same problem. On IOS, writing without response always fails because device send error code. android works well with or without response. I want to know what is the problem too. My device accepts charicteristic every 10 ms . So I changed data-sending delay but No success. My guess is that device can’t process because of too fast data delivery.

Cierpliwy, 2019-04-05:
@lim6112j Did anything changes since you commented?

lim6112j, 2019-04-05:
Device is on sale now , IOS 20 secs to send image, android 5 secs to send image now
As you know ios with feedback , android without feedback.

asteriskSF, 2019-04-29:
Is it possible the connected device does not indicate support for write without response in the characteristic attribute?  I believe iOS checks these characteristic capabilities and prevents operations that are not indicated as supported.  Sometimes devices support actions that are not properly flagged as supported...

closed 2019-07-05 by stale[bot]