dotintent/react-native-ble-plx #279: Update firmware of my ble device

biks152207, 2018-05-25:
Hey guys i have been trying to update the firmware using ble plx.But i am not sure how can i do it.I tried with writecharacteristic but i am not sure about the transaction id to do that.I have hex data of that firmware but i am very confused how we gonna inject those firmware hex data into ble device(my device is eld device, if you arenot sure about this go visit http://iosix.com/).Please suggest.

arthare, 2018-08-14:
I ended up encoding my firmware zip files into base64 strings, then sending them along as data URIs:
https://stackoverflow.com/questions/51065669/react-native-packager-configuration-how-to-include-zip-file-in-bundle

closed 2019-03-23 by stale[bot]

Irenedan, 2022-09-02:
@arthare How did you deal with it afterward?