dotintent/react-native-ble-plx #280: device.isConnected does not return the true connection state

lastmjs, 2018-05-25:
The following code prints incorrect information as far as I can tell:

```javascript
console.log('await device.isConnected()', await device.isConnected()); // logs false
await device.connect();
console.log('await device.isConnected()', await device.isConnected()); // logs false
```

The first log statement should be false, but the second log statement should be true.

lastmjs, 2018-05-25:
I think I see what's happening, the `device.isConnected()` call is returning an updated device, but the state of the original device does not update. Though I love functional programming, this is very confusing as the API is object oriented, calling a method on the object. I would expect the internal state of the device to be changing because of this. Either go fully functional or fully OO, but mixing them like this is causing confusion.

lastmjs, 2018-05-25:
Actually, this gives incorrect results as well, I would expect the bleManager to be maintaining the global state of the device at all times:

```javascript
console.log('await bleManager.isDeviceConnected(device.id)', await bleManager.isDeviceConnected(device.id));
await device.connect();
console.log('await bleManager.isDeviceConnected(device.id)', await bleManager.isDeviceConnected(device.id));
```

radko93, 2018-07-19:
It happens because `isConnected` actually uses `isDeviceConnected` under the hood https://github.com/Polidea/react-native-ble-plx/blob/484501102ed87ba4004b8a128d86245018577117/src/Device.js#L141

closed 2019-03-23 by stale[bot]

federicomiralles, 2020-04-23:
This issue is still present on version **2.0.0**. Everytime I try to check the status of a connected device by using [isDeviceConnected](https://polidea.github.io/react-native-ble-plx/#blemanagerisdeviceconnected) it always returns false.
Please check the following code, despite of sending a connected device Id, I always get **_Device AA:BB:CC:DD:EE:FF is NOT connected!_**:
```
const isBleDeviceConnected = async (deviceId = null) => {
    if (!deviceId) return false;
    const isConnected = await bleManager.isDeviceConnected(deviceId);
    if (isConnected) {
      console.log(`Device ${deviceId} is connected!`);
    } else {
      console.log(`Device ${deviceId} is NOT connected!`);
    }
    return isConnected;
  };
```

**Context:** I need to check by using the built-in RN AppState component whether a device the user was using before switching of apps (or after coming back from background) it's still connected... 

Is there any fix in progress or a workaround available?

Thanks in advance!

Federico

ad-boschung, 2020-07-20:
It seems that this issue is still in the version 2.0.1. Any news about that ? @dariuszseweryn

dariuszseweryn, 2020-07-20:
Current issue: #655