dotintent/react-native-ble-plx #281: Monitoring doesn't update/listen on iOS

Couiljer, 2018-05-30:
Hi, 

I work on a simple RN project and I'm facing an issue only on iOS. 
When I use `device.monitorCharacteristicForService()` I listen a value updated every 30s by the device. 
Everything works fine on Android, however, a strange behavior appeared on iOS : 

- on a fresh app install and first BLE connection everything is alright. 

- After the first `cancelDeviceConnection` I'm able to connect again and receive the first value returned by `device.monitorCharacteristicForService()`... but that's it. It only returns the first value that never updates. 

Any idea on this?

kevando, 2018-10-06:
I am currently struggling with the same issue. If I characteristic.read() before monitoring then I get the monitor listener function to return something, and it returns correctly. 

But if I need to read the data when I want to monitor the data, seems like I am not understanding something critical here.

Cierpliwy, 2018-10-08:
Let me clarify things a bit. Due to differences between platforms Android has 2 separate callbacks for characteristic read/characteristic notification where iOS has only one. 

Function `monitorCharacteristicForDevice` should emit only values which originated from notification/indication, which happens naturally on Android. On iOS I need to emulate this behavior by remembering how many read operations are pending and when new value arrives it is assumed to be a result of reading operation if pending read count is bigger then 0. Otherwise, the value is emitted to `monitorCharacteristicForDevice` callback.

Please send me iOS logs with invalid behavior, so I can debug it. I'll look at current code one more time to spot any issues with it.

kevando, 2018-10-08:
Thanks @Cierpliwy I am working on this today, so I will send you some error outputs.  I also started using redux-saga and it's been a huge help so thanks for that suggestion.

kevando, 2018-10-08:
I'll do my best to explain: 

I have a service with a writeable characteristic ID and a readable characteristic NAME. 

When the user writes a new ID, the NAME changes, but I am unable to catch the NAME change unless I read the characteristic after I write ID.

``` javascript
export function* setupNotifications() {
    const { connection: {device} } = yield select();
    const serviceUUID = getUUID("0000");
    const monitorCharacteristicUUID = getUUID("0003"); // NAME
    const writeCharacteristicUUID = getUUID("0001"); // ID
    const encodedString = new Buffer.from([2], 'base64').toString('base64');

    // READ
    device.readCharacteristicForService(serviceUUID,monitorCharacteristicUUID)
      .then(characteristic => {console.log('First Read:',characteristic.value)});

    //  MONITOR
    device.monitorCharacteristicForService(serviceUUID,monitorCharacteristicUUID,(error, characteristic) => {
      console.log('Monitor:',characteristic.value)});

    // mimic user interaction
    yield delay(2000);

    // WRITE NEW DATA
    yield device.writeCharacteristicWithResponseForService(serviceUUID,writeCharacteristicUUID, encodedString);

    console.log('finished writing, now check value');

    // READ AGAIN
    device.readCharacteristicForService(serviceUUID,monitorCharacteristicUUID)
      .then(characteristic => { console.log('Second Read:',characteristic.value) });
}
```

Javascript console
```
First Read: QkxVRVRPT1RILTAxAAAAAAA=
Monitor: QkxVRVRPT1RILTAxAAAAAAA=
finished writing, now check value
Second Read: R0FVR0UtMDEAAAAAAAAAAAA=
Monitor: R0FVR0UtMDEAAAAAAAAAAAA=
```

iOS Logs

```
[RxBLEKit|DEBG|18:39:25.135]: Peripheral(uuid: 33E7678F-6207-9EFC-46AB-6A21FD69162F, name: Optional("Nordic_UART")) readValue(
for: Characteristic(uuid: 94C60003-EFDB-4CF8-AC73-C4F2DCF521D8, id: 10746311040))
[RxBLEKit|DEBG|18:39:25.136]: Peripheral(uuid: 33E7678F-6207-9EFC-46AB-6A21FD69162F, name: Optional("Nordic_UART")) setNotifyValue(
enabled: true,
for: Characteristic(uuid: 94C60003-EFDB-4CF8-AC73-C4F2DCF521D8, id: 10746311040))
[RxBLEKit|DEBG|18:39:25.198]: Peripheral(uuid: 33E7678F-6207-9EFC-46AB-6A21FD69162F, name: Optional("Nordic_UART")) didUpdateValueFor(for:Characteristic(uuid: 94C60003-EFDB-4CF8-AC73-C4F2DCF521D8, id: 10746311040),
value: Optional("424c5545544f4f54482d30310000000000"),
error: nil)
[RxBLEKit|DEBG|18:39:25.258]: Peripheral(uuid: 33E7678F-6207-9EFC-46AB-6A21FD69162F, name: Optional("Nordic_UART")) didUpdateNotificationStateFor(
for:Characteristic(uuid: 94C60003-EFDB-4CF8-AC73-C4F2DCF521D8, id: 10746311040), isNotifying: true,
error: nil)
[RxBLEKit|DEBG|18:39:27.137]: Peripheral(uuid: 33E7678F-6207-9EFC-46AB-6A21FD69162F, name: Optional("Nordic_UART")) writeValue(
data: 01,
for: Characteristic(uuid: 94C60001-EFDB-4CF8-AC73-C4F2DCF521D8, id: 10746311424),
type: withResponse)
[RxBLEKit|DEBG|18:39:27.178]: Peripheral(uuid: 33E7678F-6207-9EFC-46AB-6A21FD69162F, name: Optional("Nordic_UART")) didWriteValueFor(for:Characteristic(uuid: 94C60001-EFDB-4CF8-AC73-C4F2DCF521D8, id: 10746311424),
value: Optional("02"),
error: nil)
[RxBLEKit|DEBG|18:39:27.193]: Peripheral(uuid: 33E7678F-6207-9EFC-46AB-6A21FD69162F, name: Optional("Nordic_UART")) readValue(
for: Characteristic(uuid: 94C60003-EFDB-4CF8-AC73-C4F2DCF521D8, id: 10746311040))
[RxBLEKit|DEBG|18:39:27.238]: Peripheral(uuid: 33E7678F-6207-9EFC-46AB-6A21FD69162F, name: Optional("Nordic_UART")) didUpdateValueFor(for:Characteristic(uuid: 94C60003-EFDB-4CF8-AC73-C4F2DCF521D8, id: 10746311040),
value: Optional("47415547452d3031000000000000000000"),
error: nil)
```

Debug for READ characteristic
``` javascript
deviceID: "33E7678F-6207-9EFC-46AB-6A21FD69162F"
id: 10790321536
isIndicatable: false
isNotifiable: true
isNotifying: false
isReadable: true
isWritableWithResponse: true
isWritableWithoutResponse: false
serviceID: 10750045120
serviceUUID: "94c60000-efdb-4cf8-ac73-c4f2dcf521d8"
uuid: "94c60003-efdb-4cf8-ac73-c4f2dcf521d8"
value: "QkxVRVRPT1RILTAxAAAAAAA="
```

kevando, 2018-10-09:
@Cierpliwy so is there a more appropriate way to read values then? if im reading a characteristic when the monitor event fires then it sounds like i wont capture data in that monitor event.

Is there a way to see how my pending read count?

Cierpliwy, 2018-10-09:
Looking at iOS logs in simpler form:
1) readValue(94C60003) <- you started read operation
2) setNotifyValue(94C60003) <- waiting for notifications to be registered
3) didUpdateValueFor(94C60003) <- result of read operation
4) didUpdateNotificationStateFor(94C60003) <- you can now receive notifications
5) writeValue(94C60001) <- you changed ID
6) didWriteValueFor(94C60001) <- ACK from previous operation.
7) readValue(94C60003) <- new read operation
8) didUpdateValueFor(94C60003) <- read result
this should translate to following logs on your JS side:
```
First Read: XXX
finished writing, now check value
Second Read: YYY
```
You shouldn't get Monitor logs at all, as I didn't spot any `didUpdateValueFor` calls without previous `read` operation in progress. Also note that between point 5) and 7) there is only 56 ms delay. Maybe if you could put there a delay we could see any notification?

closed 2019-03-23 by stale[bot]