dotintent/react-native-ble-plx #283: Bluetooth is not supported

amit2080799, 2018-06-04:
When I start device scan by calling the method startDevcieScan(), most of the time "bluetooth is not supported" alert comes. This happens only in ios. I am unable to figure out why its happening.

twhitacre, 2018-06-08:
@amit2080799 are you running your code on an actual device or on a simulator?

Mahaswami, 2018-06-18:
on actual device. This is the issue on all versions of ios.

amit2080799, 2018-06-18:
On actual device.

amit2080799, 2018-06-18:
the above code will work some times but most of the time it will throw alert saying "Bluetooth is not supported."

twhitacre, 2018-06-18:
That's strange. I guess it could be the device you are using. 

I've been using [this library](https://github.com/nearit/react-native-connectivity-status) to check if Bluetooth is enabled and working, so I can alert the user before I try to connect. Maybe try that and see if your device is reporting that bluetooth is in fact working.

amit2080799, 2018-06-19:
I tried in 2 3 devices which includes iphone 5s, 6s and an ipad.

pvsong, 2018-07-16:
Same problem

ioss, 2018-08-22:
Maybe a bit late, but did you check first, if the state is 'PoweredOn'? (you can request the current state, and also listen to state changes)
How soon after your app started is the above code run?

closed 2019-03-23 by stale[bot]