dotintent/react-native-ble-plx #285: Attempt to invoke virtual method on a null object reference

LucasGarci, 2018-06-11:
When i do:
```javascript
 manager = new BleManager()

    await manager.connectToDevice(props.device.id).then(device => {
      console.log({ deviceFromConnection: device })
      device
        .services()
        .then(services => console.log({ devicesServices: services }))
    })
```
i get the next error. How can i solve it? I need to view services and characteristics to overwrite some of then.

Attempt to invoke virtual method 'void com.polidea.reactnativeble.errors.BleError.reject(com.facebook.react.bridge.Promise)' on a null object reference
getServicesOrReject
    BleModule.java:1190
servicesForDevice
    BleModule.java:701
invoke
    Method.java
invoke
    JavaMethodWrapper.java:372
invoke
    JavaModuleWrapper.java:160
run
    NativeRunnable.java
handleCallback
    Handler.java:761
dispatchMessage
    Handler.java:98
dispatchMessage
    MessageQueueThreadHandler.java:29
loop
    Looper.java:156
run
    MessageQueueThreadImpl.java:192
run
    Thread.java:776

javierrebollo, 2018-06-26:
I have the same error

dariuszseweryn, 2018-06-27:
Parhaps you use JDK9 to compile Android?

javierrebollo, 2018-06-27:
Nop, I'm using java 1.8.0_172

dariuszseweryn, 2018-06-27:
And what version of the library? Could you add more context?

mtumiati, 2018-07-21:
I solved the problem of the promise reject on a null object reference adding the BleError return value in the BleErrorUtils.java file as below
```

public static BleError deviceServicesNotDiscovered(String deviceID) {
        BleError bleError = new BleError(BleErrorCode.ServicesDiscoveryFailed, null, null);
        bleError.deviceID = deviceID;
        return bleError;
    }
```
Now I'm trying to understand why it doesn't find any service. I'm using Android 8 on a Samsung S7 device. I already tried with a plugin for cordova on the same device and everything seems to be ok, so I suppose that it's not related to android version.

Cierpliwy, 2018-07-26:
@mtumiati 

Thanks for pointing that bug! I'll fix it in upcoming version. 

This error should be BleErrorCode.ServicesNotDiscovered, which means that you used function from the API, which requires disoverAllCharacteristicsAndServices() to be called successfully before usage.

closed 2019-03-23 by stale[bot]