dotintent/react-native-ble-plx #287: react-native link does not work for iOS

ethanneff, 2018-06-13:
Install manually to prevent "Native Module cannot be null" error.

ianmcgregor, 2018-06-14:
```
ld: warning: Auto-Linking framework not found BleClientManager
Undefined symbols for architecture x86_64:
  "_OBJC_CLASS_$__TtC16BleClientManager16BleClientManager", referenced from:
      objc-class-ref in libBleClient.a(BleClient.o)
  "_OBJC_CLASS_$__TtC16BleClientManager8BleEvent", referenced from:
      objc-class-ref in libBleClient.a(BleClient.o)
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
```

ericschaal, 2018-06-27:
Hey, be sure to add 
"$(SRCROOT)/../node_modules/react-native-ble-plx/ios/BleClientManager/Carthage/Build/iOS"

to **framework** search path and not header search path.

Steven-Jia, 2018-08-03:
Hey guys, I've been stuck on this for a while and would like some help.

I upgraded react-native-ble-plx to version 0.10.0 and followed the new instructions on the README, but am now getting this error on iOS for an Expo app. Linking seems to have worked properly and apparently Carthage is not needed anymore, so I deleted the framework search path in my Build Settings, set 'Always Embed Swift Standard Libraries' to 'No', and removed the run script in Build Phases.

Any ideas as to why I'm getting this error? Did I misstep somewhere?

Cierpliwy, 2018-08-03:
By this error you mean: "Native Module cannot be null"?  There is example app using version 0.10.0: https://github.com/Cierpliwy/SensorTag. Please check if it's working for you and try to spot any differences in configuration.

Steven-Jia, 2018-08-03:
Yes, I mean that error. Just checked the example app and didn't see any config differences except that 'Always Embed Swift Standard Libraries' was set to 'Yes'. I set that back to 'Yes' in my app but still get the same error.

kylewayneluck, 2018-08-09:
I'm also experiencing this issue after following your Expo/Podfile [instructions](https://github.com/Polidea/react-native-ble-plx#ios-expopodfile) exactly. Took a cursory glance at https://github.com/Cierpliwy/SensorTag provided by @Cierpliwy (above) and noticed right away that the `react-native` package is pointing towards version "0.56.0" instead of the Expo SDK fork (required in the docs provided [here](https://docs.expo.io/versions/latest/expokit/expokit#your-projects-packagejson-should-contain-a-react-native)). 

From what I understand, a detached Expo project won't run in Expo w/o that fork.

Cierpliwy, 2018-08-09:
SensorTag example is for pure react-native. I'll try to prepare version for Expo as well.

kylewayneluck, 2018-08-09:
Thanks @Cierpliwy! In the meantime, I did find a workaround for iOS using Expo. For now, if you need to test this library on an iPhone you can: 1) start the project w/ Expo SDK like normal, 2) once a tunnel connection is established in Expo, build to a connected iPhone device w/ Xcode (I followed the steps outlined in [this](https://blog.ionicframework.com/deploying-to-a-device-without-an-apple-developer-account/) article). 

It's not especially convenient, but it does get rid of the "Native module cannot be null" error.

kinjalravel, 2018-08-25:
@ianmcgregor Have you found any solution or workaround? I am getting the exact same error

askielboe, 2018-10-04:
had a similar error after updating to Xcode 10 and removing carthage. fixed it by relinking:

```
$ react-native unlink react-native-ble-plx
rnpm-install info Unlinking react-native-ble-plx ios dependency
rnpm-install info Platform 'ios' module undefined has been successfully unlinked
rnpm-install info Unlinking react-native-ble-plx android dependency
rnpm-install info Platform 'android' module undefined has been successfully unlinked
$ react-native link react-native-ble-plx
rnpm-install info Linking react-native-ble-plx ios dependency
rnpm-install info Platform 'ios' module react-native-ble-plx has been successfully linked
rnpm-install info Linking react-native-ble-plx android dependency
rnpm-install info Platform 'android' module react-native-ble-plx has been successfully linked
```

philipaarseth, 2019-02-04:
I'm getting the same error: "invariant violation: native module cannot be null"

**Using create-react-native-app**, not EXPO

What I've done so far following the installation steps:

1. `npm install --save react-native-ble-plx`
2. `react-native link react-native-ble-plx`
3. Add empty Swift file if you don't have at least one:
4. Select File/New/File...
5. Choose Swift file and click Next.
6. Name it however you want, select your targets and create it.
7. Accept to create Objective-C bridging header.
8. Minimal supported version of iOS is 8.0
9. If you want to support background mode:
10. In your application target go to Capabilities tab and enable Uses Bluetooth LE Accessories in Background Modes section.

11. Pass restoreStateIdentifier and restoreStateFunction to BleManager constructor. ( NOT DONE)

and added this to a .js file
`import { BleManager } from 'react-native-ble-plx';`

```
constructor(props) {
  	super(props);
  	this.state = {
  		text: ''
	};
	this.manager = new BleManager();
  }
```

**Tried to fix the problem:**
relinking, like @askielboe suggested:
`react-native unlink react-native-ble-plx`
`react-native link react-native-ble-plx`

restarting Xcode & bundler
I (briefly) looked at the example, but it looks like typescript, and I have no experience with TS.

philipaarseth, 2019-02-04:
May have fixed the problem, though more testing is needed.

What I did was navigating to /node_modules/react-native-ble-plx/ios/BleClient.xcodeproj and 
1. I pulled that into the Libraries folder in the project navigator, 
2. then I went to my target -> build phases -> link binary with libraries and added libBleClient.a

Cierpliwy, 2019-04-05:
Closing, as there are example apps mentioned on the README page and there is CI integration, which tests if library works properly for all supported RN versions.

closed 2019-04-05 by Cierpliwy

bukatea, 2021-01-09:
@kylewayneluck Hey, I'm running this on Expo as well, and I was running into a similar situation that you did. I have an app right now and I followed the instructions exactly on the README to get the ios setup. For me, whenever I run the app on the simulator provided by xcode, I get no such error "Native module cannot be null". However, as soon as I try to download the js bundle in the Expo ios app (following instructions laid out in [this](https://docs.expo.io/expokit/expokit) where both expo start and the xcode app is running), I get the native module error. I was wondering if this is what you were experiencing at the time of your writing the quoted post (just to make sure that we are experiencing the same situation to begin with), and whether you found any alternatives to building to a connected iPhone device instead of letting expo handle the updating itself (especially because [this](https://docs.expo.io/expokit/expokit) also mentions that expo **should** handle the updating of the js bundle. 

Thanks much!

> Thanks @Cierpliwy! In the meantime, I did find a workaround for iOS using Expo. For now, if you need to test this library on an iPhone you can: 1) start the project w/ Expo SDK like normal, 2) once [...]