dotintent/react-native-ble-plx #289: readCharacteristicForServices returns garbage value

pulkonet, 2018-06-14:
I am trying to handshake with BLE peripheral and I am writing characteristic with XXX and it must return YYY on read characteristic, but it gives garbage value.

joshua-turner, 2018-06-16:
I can't be positive without you posting some code, but it sounds like might not be taking base64 encoding into account.

pulkonet, 2018-06-16:
``sendMsg = (device) => {
		console.log('sending msg..');
		const valueBase64 = new Buffer('*BT?#').toString('base64');
		console.log(valueBase64);
		this.manager
			.writeCharacteristicWithResponseForDevice(device.id, SUUID, WCUUID, valueBase64)
			.then((res) => {
				console.log('send ', res);
				return res;
			})
			.then((res) => this.readMsg(device))
			.catch((err) => console.log(err));
	};

	readMsg = (device) => {
		console.log('reading msg..');
		this.manager
			.readCharacteristicForDevice(device.id, SUUID, RCUUID)
			.then((res) => {
				console.log('read', res, Buffer.from(res.value, 'base64').toString('ascii'));
			})
			.catch((err) => console.log(err));
	};

pulkonet, 2018-06-16:
```
sendMsg = (device) => {
		console.log('sending msg..');
		const valueBase64 = new Buffer('*BT?#').toString('base64');
		console.log(valueBase64);
		this.manager
			.writeCharacteristicWithResponseForDevice(device.id, SUUID, WCUUID, valueBase64)
			.then((res) => {
				console.log('send ', res);
				return res;
			})
			.then((res) => this.readMsg(device))
			.catch((err) => console.log(err));
	};

	readMsg = (device) => {
		console.log('reading msg..');
		this.manager
			.readCharacteristicForDevice(device.id, SUUID, RCUUID)
			.then((res) => {
				console.log('read', res, Buffer.from(res.value, 'base64').toString('ascii'));
			})
			.catch((err) => console.log(err));
	};

pulkonet, 2018-06-16:
@joshua-turner These are the two methods I am using for reading and writing. I have taken account of base64. This error might be really obvious, but I can not get it working.

pulkonet, 2018-06-18:
response from `writeCharacteristicWithResponseForDevice` promise
```
_manager: Object { _eventEmitter: {…}, _uniqueId: 9, _activePromises: {}, … }
deviceID: "D8:80:39:F4:00:0F"
id: 18
isIndicatable: false
isNotifiable: false
isNotifying: false
isReadable: false
isWritableWithResponse: true
isWritableWithoutResponse: true
serviceID: 48
serviceUUID: "49535343-fe7d-4ae5-8fa9-9fafd205e455"
uuid: "49535343-8841-43f4-a8d4-ecbe34729bb3"
value: "KkJUPyM="
__proto__: Object { … } 
```
response from `readCharacteristicForDevice` promise
```
 _manager: Object { _eventEmitter: {…}, _uniqueId: 11, _activePromises: {}, … }
deviceID: "D8:80:39:F4:00:0F"
id: 17
isIndicatable: true
isNotifiable: true
isNotifying: false
isReadable: true
isWritableWithResponse: true
isWritableWithoutResponse: true
serviceID: 48
serviceUUID: "49535343-fe7d-4ae5-8fa9-9fafd205e455"
uuid: "49535343-1e4d-4bd9-ba61-23c647249616"
value: "AAAAAAA="
```
@joshua-turner These are the console logs. If they can help.

dariuszseweryn, 2018-06-18:
The above logs look totally legit. You can check on https://cryptii.com/base64-to-hex by using
`text` => `Base64 Decode` => `bytes`
`KkJUPyM=` == `2a 42 54 3f 23`
`AAAAAAA=` == `00 00 00 00 00`
According to the above mentioned site

pulkonet, 2018-06-18:
But `AAAAAAA=` isn't what we are expecting.

dariuszseweryn, 2018-06-18:
Sorry to hear that. I am pretty sure that this is what the peripheral is responding with for a `read` request. It does not seem to be a problem with the library.

Maybe you want to check for the peripheral's response for a `write` request via characteristic notification/indication. You may want to check with the producer of your peripheral for what is the intended way of interaction.

Best Regards

pulkonet, 2018-06-18:
We have tested the peripheral with other apps (such as Bluetooth Terminal) and it seems to be working as predicted. I think I'm making some fundamental mistakes -- which unluckily I'm unable to catch -- that are hindering my development. Please look if you can help me with this. 
Thanks for support

agamb, 2018-06-18:
Hey Dariusz, I was just testing the peripheral using Sniffator and it seems to be able to read the response correctly after I turn on notify using the switch. What function do I need to call to turn on notify for my characteristic ?

dariuszseweryn, 2018-06-18:
I encourage you to read the Readme file which has a [About this library paragraph](https://github.com/Polidea/react-native-ble-plx#about-this-library) which in turn points to a [related Wiki page](https://github.com/Polidea/react-native-ble-plx/wiki/Characteristic-Notifying)

I am happy that you have resolved your problem.

Best Regards