dotintent/react-native-ble-plx #292: Crash on Android during scan

alfacommunication-alessandro, 2018-06-19:
Hi.
Start a device scan make my application always crash with following stacktrace.

My test device is an Huawei Y6 2018 with Android 8.

``` java.lang.NoSuchMethodError: No virtual method position(I)Ljava/nio/ByteBuffer; in class Ljava/nio/ByteBuffer; or its super classes (declaration of 'java.nio.ByteBuffer' appears in /system/framework/core-oj.jar)
        at com.polidea.reactnativeble.advertisement.AdvertisementData.parseScanResponseData(AdvertisementData.java:57)
        at com.polidea.reactnativeble.converter.RxBleScanResultConverter.toJSObject(RxBleScanResultConverter.java:43)
        at com.polidea.reactnativeble.converter.RxBleScanResultConverter.toJSObject(RxBleScanResultConverter.java:17)
        at com.polidea.reactnativeble.converter.JSObjectConverter.toJSCallback(JSObjectConverter.java:14)
        at com.polidea.reactnativeble.BleModule$3.call(BleModule.java:294)
        at com.polidea.reactnativeble.BleModule$3.call(BleModule.java:287)
        at rx.internal.util.ActionSubscriber.onNext(ActionSubscriber.java:39)
        at rx.observers.SafeSubscriber.onNext(SafeSubscriber.java:134)
        at rx.observers.Subscribers$5.onNext(Subscribers.java:235)
        at rx.internal.operators.OnSubscribeRefCount$2.onNext(OnSubscribeRefCount.java:120)
        at rx.internal.operators.OperatorPublish$PublishSubscriber.dispatch(OperatorPublish.java:585)
        at rx.internal.operators.OperatorPublish$PublishSubscriber.onNext(OperatorPublish.java:283)
        at rx.internal.operators.OnSubscribeMap$MapSubscriber.onNext(OnSubscribeMap.java:77)
        at rx.internal.operators.OperatorMerge$MergeSubscriber.emitScalar(OperatorMerge.java:395)
        at rx.internal.operators.OperatorMerge$MergeSubscriber.tryEmit(OperatorMerge.java:355)
        at rx.internal.operators.OperatorMerge$InnerSubscriber.onNext(OperatorMerge.java:846)
        at rx.observers.Subscribers$5.onNext(Subscribers.java:235)
        at rx.internal.operators.OnSubscribeCreate$NoneEmitter.onNext(OnSubscribeCreate.java:173)
        at rx.internal.util.ObserverSubscriber.onNext(ObserverSubscriber.java:34)
        at rx.observers.SafeSubscriber.onNext(SafeSubscriber.java:134)
        at rx.internal.operators.OperatorUnsubscribeOn$1.onNext(OperatorUnsubscribeOn.java:52)
        at rx.internal.operators.OperatorSubscribeOn$SubscribeOnSubscriber.onNext(OperatorSubscribeOn.java:74)
        at rx.internal.operators.OnSubscribeCreate$NoneEmitter.onNext(OnSubscribeCreate.java:173)
        at com.polidea.rxandroidble.internal.operations.LegacyScanOperation$1.onLeScan(LegacyScanOperation.java:40)
        at android.bluetooth.BluetoothAdapter$2.onScanResult(BluetoothAdapter.java:2540)
        at android.bluetooth.le.BluetoothLeScanner$BleScanCallbackWrapper$1.run(BluetoothLeScanner.java:535)
        at android.os.Handler.handleCallback(Handler.java:808)
        at android.os.Handler.dispatchMessage(Handler.java:101)
        at android.os.Looper.loop(Looper.java:166)
        at android.app.ActivityThread.main(ActivityThread.java:7425)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.Zygote$MethodAndArgsCaller.run(Zygote.java:245)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:921)
``` 

Environment

```
  OS: macOS High Sierra 10.13.5
  Node: 9.11.1
  Yarn: 1.7.0
  npm: 5.6.0
  Watchman: 4.9.0
  Xcode: Xcode 9.4 Build version 9F1027a
  Android Studio: 3.1 AI-173.4819257

Packages: (wanted => installed)
  react: ^16.3.2 => 16.4.1
  react-native: 0.55.4 => 0.55.4
```

dariuszseweryn, 2018-06-19:
What JDK you used to build the app? JDK9?

alfacommunication-alessandro, 2018-06-20:
> What JDK you used to build the app? JDK9?

Yes, switch back to jdk8 resolve the issue.
Thanks

closed 2018-06-20 by alfacommunication-alessandro

gre, 2018-11-28:
falling into this as well, is there a way we can make the code works everywhere?

similar crash here and it was suggested to use casting: https://github.com/plasma-umass/doppio/issues/497#issuecomment-334740243