dotintent/react-native-ble-plx #293: Unable to find any UUIDs from services and characteristics

rottenoats, 2018-06-23:
My device after calling `device.discoverAllServicesAndCharacteristics();` returns:

`{ _40: 0, _65: 0, _55: null, _72: null }`

and when I log `device` I get:

```
06-23 16:36:54.758  4397  4597 I ReactNativeJS: { isConnectable: null,
06-23 16:36:54.758  4397  4597 I ReactNativeJS:   rssi: null,
06-23 16:36:54.758  4397  4597 I ReactNativeJS:   serviceData: null,
06-23 16:36:54.758  4397  4597 I ReactNativeJS:   overflowServiceUUIDs: null,
06-23 16:36:54.758  4397  4597 I ReactNativeJS:   mtu: 23,
06-23 16:36:54.758  4397  4597 I ReactNativeJS:   localName: null,
06-23 16:36:54.758  4397  4597 I ReactNativeJS:   manufacturerData: null,
06-23 16:36:54.758  4397  4597 I ReactNativeJS:   txPowerLevel: null,
06-23 16:36:54.758  4397  4597 I ReactNativeJS:   id: '00:1E:C0:65:D7:DB',
06-23 16:36:54.758  4397  4597 I ReactNativeJS:   solicitedServiceUUIDs: null,
06-23 16:36:54.758  4397  4597 I ReactNativeJS:   name: 'RND7DB',
06-23 16:36:54.758  4397  4597 I ReactNativeJS:   serviceUUIDs: null,
06-23 16:36:54.758  4397  4597 I ReactNativeJS:   _manager:
06-23 16:36:54.758  4397  4597 I ReactNativeJS:    { _eventEmitter:
06-23 16:36:54.758  4397  4597 I ReactNativeJS:       { _subscriber:
06-23 16:36:54.758  4397  4597 I ReactNativeJS:          { _subscriptionsForType:
06-23 16:36:54.758  4397  4597 I ReactNativeJS:             { appStateDidChange:
06-23 16:36:54.758  4397  4597 I ReactNativeJS:                [ { subscriber: [Circular],
06-23 16:36:54.758  4397  4597 I ReactNativeJS:                    emitter:
06-23 16:36:54.758  4397  4597 I ReactNativeJS:                     { _subscriber: [Circular],
06-23 16:36:54.758  4397  4597 I ReactNativeJS:                       isAvailable: true,
06-23 16:36:54.758  4397  4597 I ReactNativeJS:                       _eventHandlers: { change: {}, memoryWarning: {} },
06-23 16:36:54.758  4397  4597 I ReactNativeJS:                       currentState: 'active' },
06-23 16:36:54.758  4397  4597 I ReactNativeJS:                    listener: [Function],
06-23 16:36:54.758  4397  4597 I ReactNativeJS:                    context: undefined,
06-23 16:36:54.758  4397  4597 I ReactNativeJS:                    eventType: 'appStateDidChange',
06-23 16:36:54.758  4397  4597 I ReactNativeJS:                    key: 0 } ],
06-23 16:36:54.758  4397  4597 I ReactNativeJS:               websocketMessage: [ , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , ,  ],
06-23 16:36:54.758  4397  4597 I ReactNativeJS:               websocketOpen: [ , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , ,  ],
06-23 16:36:54.758  4397  4597 I ReactNativeJS:               websocketClosed: [ , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , ,  ],
06-23 16:36:54.758  4397  4597 I ReactNativeJS:               websocketFailed: [ , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , ,  ],
06-23 16:36:54.758  4397  4597 I ReactNativeJS:               didUpdateDimensions:
06-23 16:36:54.758  4397  4597 I ReactNativeJS:                [ { subscriber: [Circular],
06-23 16:36:54.758  4397  4597 I ReactNativeJS:                    emitter:
06-23 16:36:54.758  4397  4597 I ReactNativeJS:                     { _subscriber: [Circular],
06-23 16:36:54.758  4397  4597 I ReactNativeJS:                       sharedSubscriber: [Circular],
06-23 16:36:54.758  4397  4597 I ReactNativeJS:                       _currentSubscription: null },
06-23 16:36:54.758  4397  4597 I ReactNativeJS:                    listener: [Function],
06-23 16:36:54.758  4397  4597 I ReactNativeJS:                    context: undefined,
06-23 16:36:54.758  4397  4597 I ReactNativeJS:                    eventType: 'didUpdateDimensions',
06-23 16:36:54.758  4397  4597 I ReactNativeJS:                    key: 0 } ],
06-23 16:36:54.758  4397  4597 I ReactNativeJS:               hardwareBackPress:
06-23 16:36:54.758  4397  4597 I ReactNativeJS:                [ { subscriber: [Circular],
06-23 16:36:54.758  4397  4597 I ReactNativeJS:                    emitter:
06-23 16:36:54.758  4397  4597 I ReactNativeJS:                     { _subscriber: [Circular],
06-23 16:36:54.758  4397  4597 I ReactNativeJS:                       sharedSubscriber: [Circular],
06-23 16:36:54.758  4397  4597 I ReactNativeJS:                       _currentSubscription: null },
06-23 16:36:54.758  4397  4597 I ReactNativeJS:                    listener: [Function],
06-23 16:36:54.758  4397  4597 I ReactNativeJS:                    context: undefined,
06-23 16:36:54.758  4397  4597 I ReactNativeJS:                    eventType: 'hardwareBackPress',
06-23 16:36:54.758  4397  4597 I ReactNativeJS:                    key: 0 } ],
06-23 16:36:54.758  4397  4597 I ReactNativeJS:               collectBugExtraData:
06-23 16:36:54.758  4397  4597 I ReactNativeJS:                [ { subscriber: [Circular],
06-23 16:36:54.758  4397  4597 I ReactNativeJS:                    emitter:
06-23 16:36:54.758  4397  4597 I ReactNativeJS:                     { _subscriber: [Circular],
06-23 16:36:54.758  4397  4597 I ReactNativeJS:                       sharedSubscriber: [Circular],
06-23 16:36:54.758  4397  4597 I ReactNativeJS:                       _currentSubscription: null },
06-23 16:36:54.758  4397  4597 I ReactNativeJS:                    listener: [Function: collectExtraData],
06-23 16:36:54.758  4397  4597 I ReactNativeJS:                    context: null,
06-23 16:36:54.758  4397  4597 I ReactNativeJS:                    eventType: 'collectBugExtraData',
06-23 16:36:54.758  4397  4597 I ReactNativeJS:                    key: 0 } ],
06-23 16:36:54.758  4397  4597 I ReactNativeJS:               collectRedBoxExtraData:
06-23 16:36:54.758  4397  4597 I ReactNativeJS:                [ { subscriber: [Circular],
[... 14 lines cut ...]
06-23 16:36:54.758  4397  4597 I ReactNativeJS:                    key: 0 },
06-23 16:36:54.758  4397  4597 I ReactNativeJS:                  { subscriber: [Circular],
06-23 16:36:54.758  4397  4597 I ReactNativeJS:                    emitter: { _subscriber: [Circular] },
06-23 16:36:54.758  4397  4597 I ReactNativeJS:                    listener: [Function],
06-23 16:36:54.758  4397  4597 I ReactNativeJS:                    context: undefined,
06-23 16:36:54.758  4397  4597 I ReactNativeJS:                    eventType: 'url',
06-23 16:36:54.758  4397  4597 I ReactNativeJS:                    key: 1 } ],
06-23 16:36:54.758  4397  4597 I ReactNativeJS:               toggleElementInspector:
06-23 16:36:54.758  4397  4597 I ReactNativeJS:                [ { subscriber: [Circular],
06-23 16:36:54.758  4397  4597 I ReactNativeJS:                    emitter:
06-23 16:36:54.758  4397  4597 I ReactNativeJS:                     { _subscriber: [Circular],
06-23 16:36:54.758  4397  4597 I ReactNativeJS:                       sharedSubscriber: [Circular],
06-23 16:36:54.758  4397  4597 I ReactNativeJS:                       _currentSubscription: null },
06-23 16:36:54.758  4397  4597 I ReactNativeJS:                    listener: [Function],
06-23 16:36:54.758  4397  4597 I ReactNativeJS:                    context: undefined,
06-23 16:36:54.758  4397  4597 I ReactNativeJS:                    eventType: 'toggleElementInspector',
06-23 16:36:54.758  4397  4597 I ReactNativeJS:                    key: 0 },
06-23 16:36:54.758  4397  4597 I ReactNativeJS:                  { subscriber: [Circular],
06-23 16:36:54.758  4397  4597 I ReactNativeJS:                    emitter:
06-23 16:36:54.758  4397  4597 I ReactNativeJS:                     { _subscriber: [Circular],
06-23 16:36:54.758  4397  4597 I ReactNativeJS:                       sharedSubscribe
```
At this point, I do not know what to do as I have received no valuable information to further get information from a characteristic. If anyone could provide any assistance as to how I retrieve service UUIDs and characterstics, it would be of great help. Thank you.

I'm able to receive all services and characteristics when I use Ble Scanner app on android.

Hurobaki, 2018-06-23:
Hi, 

Did you try to await this method ? It returns a promise so you have to use await inside async block or then method.

rottenoats, 2018-06-23:
@Hurobaki I do, unless I'm doing it wrong.

```
@action connect = (deviceName) => {

        const device = this.devices.find(device=>device.name === deviceName);

        if(device === undefined){
            this.info = "Couldn't find device";
            return;
        }

        this.manager.stopDeviceScan();
        this.selectedDevice = device;
        this.selectedDevice
            .connect()
            .then((device) => {
                this.info = "Discovering services and characteristics";
                return device.discoverAllServicesAndCharacteristics();
            })
            .then((device) => {
                this.info = "Setting notifications";
                return this.setupNotifications(device)
            })
            .then(() => {
                this.info = "Listening..."
            }, (error) => {
                this.info = error.message;
            })
    };

 async setupNotifications(device) {
    console.log(device);
    console.log(this.manager.servicesForDevice(device.id));
}
```

Hurobaki, 2018-06-23:
Here’s how I use it, hope it could help

```
 const { updateCharacteristic } = this.props
 const connectedDevice = await manager.connectToDevice(device.id)
 const services = await connectedDevice.discoverAllServicesAndCharacteristics()
 const characteristic = await  services.getServicesAndCharacteristics(services)

getServicesAndCharacteristics(device) {
            return new Promise((resolve, reject) => {
                device.services().then(services => {
                    const characteristics = []

                    services.forEach((service, i) => {
                        service.characteristics().then(c => {
                            characteristics.push(c)

                            if (i === services.length - 1) {
                                const temp = characteristics.reduce(
                                    (acc, current) => {
                                        return [...acc, ...current]
                                    },
                                    []
                                )
                                const dialog = temp.find(
                                    characteristic =>
                                        characteristic.isWritableWithoutResponse
                                )
                                if (!dialog) {
                                    reject('No writable characteristic')
                                }
                                resolve(dialog)
                            }
                        })
                    })
                })
            })
        }
```

rottenoats, 2018-06-24:
@Hurobaki that solved the problem :) thanks a lot. You had just a little problem, it should be `this.getServicesAndCharacteristics(services)`

Using mobx this is what is working for me:

```
    connect = flow(function * (deviceName){

        const device = this.devices.find(device=>device.name === deviceName);

        if(device === undefined){
            this.info = "Couldn't find device";
            return;
        }

        this.manager.stopDeviceScan();

        const connectedDevice = yield this.manager.connectToDevice(device.id);
        const services =        yield connectedDevice.discoverAllServicesAndCharacteristics();
        const characteristic =  yield this.getServicesAndCharacteristics(services);

        console.log(characteristic);

    });

    getServicesAndCharacteristics(device) {
        return new Promise((resolve, reject) => {
            device.services().then(services => {
                const characteristics = [];
                services.forEach((service, i) => {
                    service.characteristics().then(c => {
                        characteristics.push(c);
                        if (i === services.length - 1) {
                            const temp = characteristics.reduce(
                                (acc, current) => {
                                    return [...acc, ...current]
                                },
                                []
                            );
                            const dialog = temp.find(
                                characteristic =>
                                    characteristic.isWritableWithoutResponse
                            );
                            if (!dialog) {
                                reject('No writable characteristic')
                            }
                            resolve(dialog)
                        }
                    })
                })
            })
        })
    }
```

closed 2018-06-24 by rottenoats

ItsHraggg, 2023-01-23:
Hi, I need help with the same problem for me. I want to connect and read/write to my ble module, although I cannot get the services and characteristics for my device. I want to know the code above, when we create getServicesAndCharacteristics(device){}, is this a function inside my App function? or it is a constant?