dotintent/react-native-ble-plx #297: [1.0] Promise interruption

Cierpliwy, 2018-06-29:
# Promise interruption

As new library is using only `Promise` based API we need to make sure that
it is possible to cancel any operation in the most convenient way. Current
implementation supports `transactionId` field which can be used to cancel
any `Promise`. Most of a time it is convenient to cancel Promises due to
timeouts or in groups. For example if there are two functions blocked on
Buffer, cancelling or closing it should cancel all pending promises 
related to it.

All APIs currently accept last `transactionId` parameter. There is one 
function to cancel every  `Promise` based on `transactionId`. However in 
such situation it is inconvenient in more complex situation:

```js
async function f {
    ...
    const transactionId = "xx"
    setTimeout(function() {
        cancel(transactionId)
    }, 1000)
    const value = await takeLatest(buffer, transactionId)
    ...
}
```

We can extend cancellation API to be an object with additional fields:

```js
type CancelOptions = {
     id?: string,
     timeout?: number,
}
```

Most common cancellation cases will be handled properly:

```js
async function f {
    ...
    const value = await takeLatest(buffer, { timeout: 1000 })
    ...
}
```

Additionally we can reuse this object and merge it with other options in other functions such as scanning:

```js
startDeviceScan(null, { allowDuplicates: true, timeout: 5000 })`
```

When promise can return reasonable value instead of an cancelled error (empty array, null value etc.) 
we can use additional option to do that:

```js
const values = await takeLatestChunk(buffer, 10, { timeout: 1000, ignoreCancelled: true })
```

Above function will return 10 values or lower number of values accumulated in 1 sec. Without
`ignoredCancelled` option it would return an error.

dariuszseweryn, 2018-06-29:
It's not clear to me how will `CancelOptions.ignoreCancelled` will work with different APIs.

Cierpliwy, 2018-06-29:
When `Cancelled` error is about to be emitted due to manual `cancel` call or timeout being triggered and function has sensible default return value, promise resolves with it. In the end, any function which will use this parameter will have note in documentation how it behaves.

closed 2019-03-23 by stale[bot]