dotintent/react-native-ble-plx #298: [1.0] API signatures

Cierpliwy, 2018-06-29:
# 1.0 API function and object list.

Following link represents all functions and objects to be implemented
before reaching 1.0 release:

* [ ] `BleError` (global namespace)
  ```js
  class BleError {
      errorCode: BleErrorCode,
      attErrorCode: ?BleATTErrorCode,
      iosErrorCode: ?BleIOSErrorCode,
      androidErrorCode: ?BleAndroidErrorCode,
      reason: ?string
  }
  ```
* [ ] `BleErrorCode` (global namespace)
* [ ] `BleATTErrorCode` (global namespace)
* [ ] `BleAndroidErrorCode` (global namespace)
* [ ] `BleIOSErrorCode` (global namespace)
* [ ] `Utils` namespace
  * Functions:
    * [ ] `to128LowerCaseUUID(uuid: UUID): UUID`
* [ ] `Promise` namespace
  * [ ] Types:
    * [ ] `type PromiseId = string`
    * [ ] CancelOptions
    ```js
    type CancelOptions = {
      promiseId?: PromiseId,
      timeout?: number
    }
    ```
  * [ ] Functions:
    * [ ] `cancel(promiseId?: PromiseId): void`
* [ ] `Buffer` namespace
  * [ ] Types:
    * [ ] Buffer
      ```js
      type Buffer<T> = {
          id: number
      }
      ```
  * [ ] Functions:
    * [ ] `takeLatest(options: ?CancelOptions): Promise<T>`
    * [ ] `takeLatestChunk(size: number, options: ?CancelOptions): Promise<T>`
    * [ ] `takeOldest(options: ?CancelOptions): Promise<T>`
    * [ ] `takeOldestChunk(size: number, options: ?CancelOptions): Promise<T>`
    * [ ] `takeAll(options: ?CancelOptions): Promise<T>`
    * [ ] `peekLatest(options: ?CancelOptions): Promise<T>`
    * [ ] `peekLatestChunk(size: number, options: ?CancelOptions): Promise<T>`
    * [ ] `peekOldest(options: ?CancelOptions): Promise<T>`
    * [ ] `peekOldestChunk(size: number, options: ?CancelOptions): Promise<T>`
    * [ ] `peekAll(options: ?CancelOptions): Promise<T>`
    * [ ] `flushLatest(options: ?CancelOptions): Promise<T>`
    * [ ] `flushLatestChunk(size: number, options: ?CancelOptions): Promise<T>`
    * [ ] `flushOldest(options: ?CancelOptions): Promise<T>`
    * [ ] `flushOldestChunk(size: number, options: ?CancelOptions): Promise<T>`
    * [ ] `flushAll(options: ?CancelOptions): Promise<T>`
    * [ ] `destroy(buffer: Buffer<T>): void`
* [ ] `CentralManager` namespace
  * [ ] Types:
    * [ ] UUID
      ```js
      type UUID = string
      ```
    * [ ] CentralManager
      ```js
      type CentralManager = {
          id: number
      }
      ```
    * [ ] CentralManagerOptions
      ```js
      type CentralManagerOptions = {
          restoreStateId?: string
      }
      ```
    * [ ] `type LogLevelType = $Keys<typeof LogLevel>`
    * [ ] `type StateType = $Keys<typeof State>`
    * [ ] Peripheral
      ```js
      type Peripheral = {
          id: string,
          centralId: number,
          name: ?string
      }
      ```
    * [ ] Service
      ```js
      type Service = {
          id: number,
          peripheralId: string,
          centralId: number,
          uuid: UUID,
          isPrimary: boolean
      }
      ```
    * [ ] Characteristic
      ```js
      type Characteristic = {
          id: number,
          serviceId: number,
          serviceUUID: UUID,
          peripheralId: string,
          centralId: number,
          uuid: UUID,
          isReadable: boolean,
          isWritableWithResponse: boolean,
          isWritableWithoutResponse: boolean,
          isNotifiable: boolean,
          isIndicatable: boolean,
          isNotifying: boolean
      }
      ```
    * [ ] ScanRecord
      ```js
      type ScanRecord = {
          peripheral: Peripheral,
          rssi: number,
          advertisement: AdvertisementData
      }
      ```
    * [ ] AdvertisementData
      ```js
      type AdvertisementData = {
          manufacturerDataBase64: ?string,
          manufacturerData: ?ArrayBuffer,
          serviceDataBase64: ?{[UUID]: string},
          serviceData: ?{[UUID]: ArrayBuffer},
          serviceUUIDs: ?Array<UUID>,
          localName: ?string,
          txPowerLevel: ?number,
          solicitedServiceUUIDs: ?Array<UUID>,
          isConnectable: ?boolean,
          overflowServiceUUIDs: ?Array<UUID>
      }
      ```
    * [ ] ScanOptions
      ```js
      type ScanOptions = {
          allowDuplicates?: boolean
      }
      ```
    * [ ] ConnectionOptions
      ```js
      type ConnectionOptions = {
          autoConnect?: boolean,
          requestMTU?: boolean,
          refreshGatt?: boolean
      }
      ```
    * [ ] RestoredState
      ```js
      type RestoredState = {
          connectedPeripherals?: Array<Peripheral>,
          scanOptions?: ScanOptions
      }
      ```
  * [ ] Objects:
      * [ ] LogLevel
      ```js
      const LogLevel = {
        None: 'None',
        Verbose: 'Verbose',
        Debug: 'Debug',
        Info: 'Info',
        Warning: 'Warning',
        Error: 'Error',
      }
      ```
      * [ ] State
      ```js
      const LogLevel = {
        Unknown: 'Unknown',
        Resetting: 'Resetting',
        Unsupported: 'Unsupported',
        Unauthorized: 'Unauthorized',
        PoweredOff: 'PoweredOff',
        PoweredOn: 'PoweredOn'
      }
      ```
  * [ ] Functions. Note: all Promise returning functions have implicit `CancelOptions` at the end 
        of their signature. If there are any function specific options, they are merged together.

    * [ ] `setLogLevel(logLevel: LogLevelType): Promise<void>`
    * [ ] `getLogLevel(): Promise<LogLevelType>`
    ---
    * [ ] `create(options: CentralManagerOptions = {}): Promise<CentralManager>`
    * [ ] `destroy(central: CentralManager): Promise<void>`
    * [ ] `getRestoredState(central: CentralManager): Promise<?RestoredState>`
    * [ ] `getState(central: CentralManager): Promise<StateType>`
    * [ ] `monitorState(central: CentralManager): Promise<Buffer<StateType>>`
    * [ ] `scanForPeripherals(central: CentralManager, serviceUUIDs: ?Array<UUID>, options: ScanOptions = {}): Promise<Buffer<ScanRecord>>`
    * [ ] `getPeripheral(central: CentralManager, peripheralId: string): Promise<Peripheral>`
    * [ ] `getPeripherals(central: CentralManager, peripheralIds: Array<string>): Promise<Array<Peripheral>>`
    * [ ] `getConnectedPeripherals(central: CentralManager, serviceUUIDs: Array<UUID>): Promise<Array<Peripheral>>`
    ---
    * [ ] `connect(peripheral: Peripheral, options: ConnectionOptions = {}): Promise<Peripheral>`
    * [ ] `cancelConnection(peripheral: Peripheral): Promise<void>`
    * [ ] `isConnected(peripheral: Peripheral): Promise<boolean>`
    * [ ] `monitorDisconnection(peripheral: Peripheral): Promise<Buffer<[Peripheral, ?BleError]>>`
    * [ ] `readRSSI(peripheral: Peripheral): Promise<number>`
    * [ ] `requestMTU(peripheral: Peripheral): Promise<number>`
    * [ ] `getMTU(peripheral: Peripheral): Promise<number>`
    * [ ] `discoverAllServicesAndCharacteristics(peripheral: Peripheral): Promise<void>`
    * [ ] `getService(peripheral: Peripheral, serviceUUID: UUID): Promise<Service>`
    * [ ] `getServices(peripheral: Peripheral): Promise<Array<Service>>`
    ---
    * [ ] `getCharacteristicForService(peripheral: Peripheral, serviceUUID: UUID, characteristicUUID: UUID): Promise<Characteristic>`
    * [ ] `getCharacteristic(service: Service, characteristicUUID: UUID): Promise<Characteristic>`
    * [ ] `getCharacteristics(service: Service): Promise<Array<Characteristic>>`
    * [ ] `readBase64CharacteristicValue(characteristic: Characteristic): Promise<string>`
    * [ ] `readCharacteristicValue(characteristic: Characteristic): Promise<ArrayBuffer>`
    * [ ] `writeBase64CharacteristicValueWithResponse(characteristic: Characteristic, base64Value: string): Promise<void>`
    * [ ] `writeCharacteristicValueWithResponse(characteristic: Characteristic, value: ArrayBuffer): Promise<void>`
    * [ ] `writeBase64CharacteristicValueWithoutResponse(characteristic: Characteristic, base64Value: string): Promise<void>`
    * [ ] `writeCharacteristicValueWithoutResponse(characteristic: Characteristic, value: ArrayBuffer): Promise<void>`
    * [ ] `monitorBase64CharacteristicValue(characteristic: UUID): Promise<Buffer<string>>`
    * [ ] `monitorCharacteristicValue(characteristic: UUID): Promise<Buffer<ArrayBuffer>>`

dariuszseweryn, 2018-06-29:
Minor issues in the post:
- probably not a `LogLevel`?
```
const LogLevel = {
  Unknown: 'Unknown',
  Resetting: 'Resetting',
  Unsupported: 'Unsupported',
  Unauthorized: 'Unauthorized',
  PoweredOff: 'PoweredOff',
  PoweredOn: 'PoweredOn'
}
```
- probably a `LogLevel` not a `LogLevelType`?
```
setLogLevel(logLevel: LogLevelType): Promise<void>
getLogLevel(): Promise<LogLevelType>
```
- shouldn't `takeAll(options: ?CancelOptions): Promise<T>` be `takeAll(options: ?CancelOptions): Promise<[T]>`?

paweljaneczek, 2018-06-29:
Proposition:

1. Instead of `ignoreCancelled` in `CancelOptions` I would propose:
- `alwaysReturnResult` - I think that is best one
- `treatTerminationAsSuccess`
- `returnResultOnFinish`

2. All buffer functions like `takeLatest(options: ?CancelOptions): Promise<T>` should have `Buffer<T>` as first argument, so e.g. `takeLatest(buffer: Buffer<[T]>, options: ?CancelOptions): Promise<[T]>`

3. I would remove for now `flush*` buffer methods and add this as a future issue if that will be needed.

4. CentralManagerOptions should have also parameter `showPowerAlert`, like:
```
type CentralManagerOptions = {
    restoreStateId?: string,
    showPowerAlert?: string
}
```
More [here](https://developer.apple.com/documentation/corebluetooth/cbcentralmanageroptionshowpoweralertkey?changes=_1)

5. In `PeripheralType` I would also add uuid, and change `id` to number, so:
```
type Peripheral = {
    id: number,
    uuid: UUID,
    centralId: number,
    name: ?string
}
```
With that change we would need also to change `ServiceType` and `CharacteristicType` - change `peripheralId` and add `peripheralUuid`.

6. Typo, change `LogLevel` to `State`:
```
const State = {
  Unknown: 'Unknown',
  Resetting: 'Resetting',
  Unsupported: 'Unsupported',
  Unauthorized: 'Unauthorized',
  PoweredOff: 'PoweredOff',
  PoweredOn: 'PoweredOn'
}
```

7. As discussed before, we want to have buffer also for restoredState so nothing will be missed. Finally it should be:
```
getRestoredState(central: CentralManager): Promise<Buffer<?RestoredState>>
```

8. `getPeripheral(central: CentralManager, peripheralId: string): Promise<Peripheral>` should be `getPeripheral(central: CentralManager, peripheralUuid: UUID): Promise<Peripheral>`. Question is do we want to alse get by `peripheralId: number`? or maybe only second option?

9. `discoverAllServicesAndCharacteristics(peripheral: Peripheral): Promise<void>` shouldn't we return `Promise<Peripheral>`? Won't it be better for usage?

Cierpliwy, 2018-06-29:
1. I would also suggest `forceResult`
2. Done
3. Will do for now as that can be added later.
4. That's right
5. Not sure about that. What about Android were there is no UUID but MAC address.
6. Fixed.
7. Not sure about that. Buffer suggests that there are multiple values that could be emitted. We could create
   buffer as implementation detail so that we won't lost restored state.
8. In case of Android we don't have UUID.
9. With `await` and `async` syntax it doesn't matter.
10. I think it should be defined in options.

paweljaneczek, 2018-06-29:
1. `forceResult` is also good 👍 I think that best would be to meet with @DariuszAniszewski during next week and choose best one.
5. Didn't know, I am just thinking that there is naming conflict now. We are using `id` as UUID/MAC and for our internal number `id`. But for now don't have any proposition how to solve it.
7. Yeah, you are right. Hidding it as implementation details is a good way :+1:
10. So how you want to handle `Options` types for each method? Will those be e.g. unions for `CancelOptions`, `BufferOptions` & `ScanningOptions` or maybe some new type?

DariuszAniszewski, 2018-06-29:
1. I guess you want to meet with @dariuszseweryn :)

dariuszseweryn, 2018-06-29:
1. Food for thought. Or maybe always return some default and have an option like `cancelIsError`?

closed 2019-03-23 by stale[bot]