dotintent/react-native-ble-plx #300: BleError device was disconnected

dominikeinkemmer, 2018-07-01:
Hi,

I have posted this previously as an error on version 0.8:
https://github.com/Polidea/react-native-ble-plx/issues/212

On version 0.9.1 I now get the following error when I cut the power supply of my bluetooth device. I am still not sure how to deal with this. Where in my project should I catch these errors and catch them so that I don't crash the app but rather shutdown gracefully and return the user to the `scan for device` screen.

[image]

Thanks in advance.

dominikeinkemmer, 2018-07-01:
In case someone else faces this:

After playing around with this a lot, I found the problem. Inside the function I used for `monitorCharacteristicForDevice` I found that I used `console.error` to log the error instead of handling this.

Should this maybe be mentioned in the docs that this function will receive the disconnect error?

closed 2018-07-01 by dominikeinkemmer

will-thrive, 2023-01-17:
> After playing around with this a lot, I found the problem. Inside the function I used for `monitorCharacteristicForDevice` I found that I used `console.error` to log the error instead of handling this.

How did you handle it exactly?  Please would you post a code snippet?