dotintent/react-native-ble-plx #302: Is there a way to monitor when a connection is established?

ericschaal, 2018-07-06:
Hi!

Is there a way to monitor when a connection is established? This is useful when `autoConnect` is enabled to detect when a connection has been reestablished.

Thank you

Cierpliwy, 2018-07-09:
According to RxAndroidBle documentation:

> Unlike the native Android API, if set to true [autoconnect] and the connection is lost there will NOT be any attempt to reconnect unless explicitly resubscribed.

So currently there is no need to monitor connection in current API. Correct me @dariuszseweryn if I'm wrong here.

dariuszseweryn, 2018-07-09:
> According to RxAndroidBle documentation:
>
>
> So currently there is no need to monitor connection in current API. Correct me @dariuszseweryn if I'm wrong here.

Correct.

closed 2019-03-23 by stale[bot]