dotintent/react-native-ble-plx #303: Update 0.8 to 0.9 value.replace is not an object

poptocrack, 2018-07-10:
After I've made the update from 0.8 to 0.9.1 an error is thrown.:

undefined is not an object (evaluation value.replace)

fillStringWithArguments in utils.js

After a couple of hours trying to figure out this issue, I couldn't manage to find anything. I didn't change anything in my part of the code, so i suspect it comes from the lib. But if I'm the only one with this issue, it might comes from me.

Any idea about this?

Cierpliwy, 2018-07-10:
This looks like a bug on my side. Can you tell me if you are using Android or iOS? Which function throws this error?

poptocrack, 2018-07-10:
I'm actually running some tests, I keep you in touch.

poptocrack, 2018-07-10:
After apr. 2h of tests, I can't reproduce the error. I'm closing this for now.

closed 2018-07-10 by poptocrack