dotintent/react-native-ble-plx #307: __abort_with_payload / _dyld_start

jason2li, 2018-07-17:
I am trying to develop an iOS (and Android) app.

It works perfectly on Android, but I constantly get the same error on iOS. I have even started a brand new react-native project from scratch in an effort to narrow it down to the lowest common denominator, and this issue happens each time I try to run it after adding and linking the react-native-ble-plx package.

[Here is a screenshot of the error](https://i.stack.imgur.com/2aCvW.png)

- I've tried clearing caches and rebooting
- I've tried rolling back to previous versions
- I've tried [embedding the 3rd party package to the embedded binaries](https://stackoverflow.com/a/42185158/462158), but I'm unsure what to add (I don't see a .framework file)

Any help would be greatly appreciated, I've already invested about 40 hours trying to figure this out, and I'm stumped.

I'm new to iOS development, so I'm not sure what information is would help, but here are the versions to several things:

- OS: maxOS High Sierra Version 10.13.6
- Xcode: Version 9.4.1 (9F2000)
- Swift: Version 4.1.2 (swiftlang-902.0.54 clang-902.0.39.2)
- react: Version 16.4.1
- react-native: Version 0.56.0
- react-native-ble-plx: Version 0.9.1
- npm: Version: 6.2.0
- Yarn: Version 1.7.0
- Carthage: Version 0.30.1

jason2li, 2018-07-18:
In case anybody else was as stumped as I was, maybe this will help.

I submitted this to [another site](https://gitter.im/RxBLELibraries/react-native-ble?source=all-rooms-list), and was graciously helped by another user there.  He created [this repo](https://github.com/Cierpliwy/SensorTag), which works perfectly.

I haven't taken the time to figure out what I did wrong just yet (time crunch), but hopefully this will help others.

paweljaneczek, 2018-07-27:
It's great that it finally works for you! Closing this issue

closed 2018-07-27 by paweljaneczek