dotintent/react-native-ble-plx #308: compiled with newer version of Swift language (unknown ABI version 0x06) than previous files

SchwarzerSchnee, 2018-07-18:
Hello. Had to update my Xcode recently for other reasons, and then this build failure happened:

ld: /Users/***/node_modules/react-native-ble-plx/ios/BleClientManager/Carthage/Build/iOS/RxBluetoothKit.framework/RxBluetoothKit compiled with newer version of Swift language (unknown ABI version 0x06) than previous files (4.0) file '/Users/***/node_modules/react-native-ble-plx/ios/BleClientManager/Carthage/Build/iOS/RxBluetoothKit.framework/RxBluetoothKit' for architecture armv7
clang: error: linker command failed with exit code 1 (use -v to see invocation)

Carthage output:

*** Cloning RxBluetoothKit
*** Cloning RxSwift
*** Checking out RxBluetoothKit at "4.0.1"
*** Checking out RxSwift at "4.2.0"
*** xcodebuild output can be found in /var/folders/j2/98qn3fqj4h92xj72xpsbwqfc0000gn/T/carthage-xcodebuild.YGLgT4.log
*** Downloading RxSwift.framework binary at "Zürich"
*** Downloading RxBluetoothKit.framework binary at "4.0.1"
***  Skipped installing RxBluetoothKit.framework binary due to the error:
	"Incompatible Swift version - framework was built with 4.0.2 (swiftlang-900.0.69.2 clang-900.0.38) and the local version is 4.1.2 (swiftlang-902.0.54 clang-902.0.39.2)."

    Falling back to building from the source
*** Building scheme "RxBluetoothKit macOS" in RxBluetoothKit.xcodeproj
*** Building scheme "RxBluetoothKit iOS" in RxBluetoothKit.xcodeproj

Versions:
Xcode 9.4.1
iOS 9-11
React Native 0.51
react-native-ble-plx 0.9.1

I have tried all sorts of things -- deleting Carthage cache, deleting DerivedData, installing other Swift toolchains, re-installing everything etc. -- to no avail, and I'm just about at my wits' end. I may have even done the right things just in the wrong order, or just missed one little detail, or whatever, I don't know.

Apparently this error doesn't even occur on a fresh project... but well, moving everything to a fresh project, let alone renaming it, isn't an option here because of stored user data. It seems there's something bad hidden deeply somewhere that can't fix itself with setting everything up anew, how do I get rid of it?

SchwarzerSchnee, 2018-07-20:
Okay, I ended up finding the solution on my own, so for all other people who might be having the same issue...

First I removed all traces of ble-plx and some other annoying stuff in my project, then added it again. Then I learned that ble-plx by itself was actually working just fine, but the described problem was really caused by react-native-nordic-dfu -- another open-source lib that apparently uses ble-plx. I edited the Cartfiles in that lib to match my current Swift version, did a `carthage update`on it, and replaced its old frameworks in the Xcode project with the new ones. That resolved this whole conflict.

paweljaneczek, 2018-07-27:
It looks like it has been resolved. Thanks @SchwarzerSchnee for solution. Closing

closed 2018-07-27 by paweljaneczek