dotintent/react-native-ble-plx #313: Example of operation

NdozHH, 2018-08-01:
Can you integrate a basic example of device scanning?

Tilwin90, 2018-09-19:
I threw an adapter that abstracts BleManager, and I included the following two methods in it:

```
startDeviceScan(scanCallback) {
    const uuids = [/*Insert UUIDs here*/];
    const scanOptions = { allowDuplicates: false };
    this.bleManager.startDeviceScan(uuids, scanOptions, scanCallback);
}

stopDeviceScan() {
  this.bleManager.stopDeviceScan();
}
```

Then as a response to a scan trigger, you can have the following code:

```
function deviceFound(error, device, dispatch) {
  if (!error) {
    // do something with device - if you use Redux, you probably want to dispatch an action here that a device has been discovered
  }
}

.....
// when scan triggered
const SCAN_DURATION = 5000;
adapter.startDeviceScan(deviceFound);
setTimeout(() => adapter.stopDeviceScan(), SCAN_DURATION);

```

Cierpliwy, 2018-10-05:
Scanning is described in documentation and present in example configuration project mentioned in README file. Closing.

closed 2018-10-05 by Cierpliwy