dotintent/react-native-ble-plx #314: installation fails: Could not find com.android.support:support-annotations:27.1.0.

axlider, 2018-08-01:
After I installed react-native-ble-plx and ran "react-native run-android" I get the following error.
Do I have to upgrade "react-native" currently (0.52.0)?
I tried adding dependency to build.gradle for "com.android.support:support-v4:23.0.1" but without success.

Thanks for any help!

```
A problem occurred configuring project ':app'.
> Could not resolve all dependencies for configuration ':app:_debugApkCopy'.
   > Could not find com.android.support:appcompat-v4:23.0.1.
     Searched in the following locations:
         file:/Users/b9179/Library/Android/sdk/extras/android/m2repository/com/android/support/appcompat-v4/23.0.1/appcompat-v4-23.0.1.pom
         file:/Users/b9179/Library/Android/sdk/extras/android/m2repository/com/android/support/appcompat-v4/23.0.1/appcompat-v4-23.0.1.jar
         file:/Users/b9179/ws/wirt/android/sdk-manager/com/android/support/appcompat-v4/23.0.1/appcompat-v4-23.0.1.jar
     Required by:
         wirt:app:unspecified
   > Could not find com.android.support:support-annotations:27.1.0.
     Searched in the following locations:
         file:/Users/b9179/Library/Android/sdk/extras/android/m2repository/com/android/support/support-annotations/27.1.0/support-annotations-27.1.0.pom
         file:/Users/b9179/Library/Android/sdk/extras/android/m2repository/com/android/support/support-annotations/27.1.0/support-annotations-27.1.0.jar
         file:/Users/b9179/ws/wirt/android/sdk-manager/com/android/support/support-annotations/27.1.0/support-annotations-27.1.0.jar
     Required by:
         wirt:app:unspecified > wirt:react-native-ble-plx:unspecified > com.polidea.rxandroidble:rxandroidble:1.6.0
   > Could not find com.android.support:support-annotations:27.1.0.
     Searched in the following locations:
         file:/Users/b9179/Library/Android/sdk/extras/android/m2repository/com/android/support/support-annotations/27.1.0/support-annotations-27.1.0.pom
         file:/Users/b9179/Library/Android/sdk/extras/android/m2repository/com/android/support/support-annotations/27.1.0/support-annotations-27.1.0.jar
         file:/Users/b9179/ws/wirt/android/sdk-manager/com/android/support/support-annotations/27.1.0/support-annotations-27.1.0.jar
     Required by:
         wirt:app:unspecified > com.android.support:appcompat-v7:23.0.1 > com.android.support:support-v4:23.0.1

```

axlider, 2018-08-08:
Update:

In build.gradle I made following change:

Instead of 
```
 maven {
            // All of React Native (JS, Obj-C sources, Android binaries) is installed from npm
            url "$rootDir/../node_modules/react-native/android"
        }
```
I added:
`maven { url 'https://maven.google.com' }`

So it looks like:
```
allprojects {
    repositories {
        mavenLocal()
        jcenter()
         maven { url 'https://maven.google.com' }
        /*maven {
            // All of React Native (JS, Obj-C sources, Android binaries) is installed from npm
            url "$rootDir/../node_modules/react-native/android"
        }*(
    }
}
```

Seems like above error has gone,
but then I get compilation errors.

:app:generateDebugSources
:app:incrementalDebugJavaCompilationSafeguard
:app:compileDebugJavaWithJavac
:app:compileDebugJavaWithJavac - is not incremental (e.g. outputs have changed, no previous execution, etc.).
/Users/b9179/ws/wirt/android/app/src/main/java/com/wirt/MainApplication.java:5: error: cannot find symbol
import com.facebook.react.ReactApplication;
                         ^
  symbol:   class ReactApplication
  location: package com.facebook.react
/Users/b9179/ws/wirt/android/app/src/main/java/com/wirt/MainApplication.java:9: error: cannot find symbol
import com.facebook.react.ReactNativeHost;
                         ^
  symbol:   class ReactNativeHost
  location: package com.facebook.react
/Users/b9179/ws/wirt/android/app/src/main/java/com/wirt/MainApplication.java:18: error: cannot find symbol
public class MainApplication extends Application implements ReactApplication {
                                                            ^
  symbol: class ReactApplication
/Users/b9179/ws/wirt/android/app/src/main/java/com/wirt/MainApplication.java:20: error: cannot find symbol
  private final ReactNativeHost mReactNativeHost = new ReactNativeHost(this) {
                ^
  symbol:   class ReactNativeHost
  location: class MainApplication
/Users/b9179/ws/wirt/android/app/src/main/java/com/wirt/MainApplication.java:44: error: cannot find symbol
  public ReactNativeHost getReactNativeHost() {
         ^
  symbol:   class ReactNativeHost
  location: class MainApplication
/Users/b9179/ws/wirt/android/app/src/main/java/com/wirt/MainActivity.java:5: error: MainActivity is not abstract and does not override abstract method getPackages() in ReactActivity
public class MainActivity extends ReactActivity {
       ^
/Users/b9179/ws/wirt/android/app/src/main/java/com/wirt/MainApplication.java:20: error: cannot find symbol
  private final ReactNativeHost mReactNativeHost = new ReactNativeHost(this) {
                                                       ^
  symbol:   class ReactNativeHost
  location: class MainApplication
/Users/b9179/ws/wirt/android/app/src/main/java/com/wirt/MainApplication.java:43: error: method does not override or implement a method from a supertype
  @Override
  ^
8 errors
:app:compileDebugJavaWithJavac FAILED

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':app:compileDebugJavaWithJavac'.
> Compilation failed; see the compiler error output for details.

axlider, 2018-08-08:
This setup fixed it:

```
allprojects {
    repositories {
        mavenLocal()
        jcenter()
        maven { url 'https://maven.google.com' }
        maven {
            // All of React Native (JS, Obj-C sources, Android binaries) is installed from npm
            url "$rootDir/../node_modules/react-native/android"
        }
    }
}
```

closed 2018-08-08 by axlider

kevando, 2018-08-15:
@axlider thank you so much for this!! How on earth did you figure out this fix?

axlider, 2018-08-17:
glad this also helped someone else.

arthare, 2018-09-27:
Alternative solution: Download the .jar and .pom from https://mvnrepository.com/artifact/com.android.support/support-annotations/27.1.0 and put them in the directory it says it is hunting for.