dotintent/react-native-ble-plx #315: Connection not completing or getting weird errors

darkrift, 2018-08-01:
I have acquired a new android device (samsung SM-T380 on android 7.1) which exhibits a weird behavior quite often actually. At some point (which I have not been able to point how to reproduce, but usually triggers with hot reload), the bluetooth connection starts to act weird, receiving various errors which prevents connection with device. 

The only way I was able to actually come back on track is to reset network settings of Android.

Here's one of the log extract with logcat when the device actually doesn't "complete" the connection : 
```
08-01 11:06:02.307  7809  8319 I ReactNativeJS: [ { [Error: Disconnected]
08-01 11:06:02.307  7809  8319 I ReactNativeJS:     line: 193325,
08-01 11:06:02.307  7809  8319 I ReactNativeJS:     column: 22,
08-01 11:06:02.307  7809  8319 I ReactNativeJS:     sourceURL: 'http://192.168.1.30:8081/index.delta?platform=android&dev=true&minify=false' } ]
08-01 11:06:08.200  1049  1599 E Watchdog: !@Sync 24 [2018-08-01 11:06:08.200]
08-01 11:06:11.443  7809  8419 D BluetoothGatt: connect() - device: 00:13:43:3B:18:XX, auto: false
08-01 11:06:11.444  7809  8419 D BluetoothGatt: registerApp()
08-01 11:06:11.445  7809  8419 D BluetoothGatt: registerApp() - UUID=bdb7b1a5-d41b-441f-92f4-16a0fec23b41
08-01 11:06:11.549  7809  7838 D BluetoothGatt: onClientRegistered() - status=0 clientIf=7
08-01 11:06:11.672  1472  1862 E bt_btif : bta_gattc_rebuild_cache: bta_gattc_rebuild_cache
08-01 11:06:11.697  1472  1571 E BluetoothRemoteDevices: aclStateChangeCallback: State:Connected to Device:00:13:43:3B:18:XX, linktype is 2
08-01 11:06:11.701  7809  7839 D BluetoothGatt: onClientConnectionState() - status=0 clientIf=7 device=00:13:43:3B:18:XX
08-01 11:06:11.732  7809  8419 D BluetoothGatt: discoverServices() - device: 00:13:43:3B:18:XX
08-01 11:06:11.738  7809  7838 D BluetoothGatt: onSearchComplete() = Device=00:13:43:3B:18:XX Status=0
08-01 11:06:11.756  7809  8320 D BluetoothGatt: setCharacteristicNotification() - uuid: 18cda784-4bd3-4370-85bb-bfed91ec86af enable: true
08-01 11:06:12.354  7809  7839 W BluetoothGatt: onCharacteristicRead() - Device=00:13:43:3B:18:XX handle=22 Status=0
08-01 11:06:12.451  7809  7838 W BluetoothGatt: onCharacteristicRead() - Device=00:13:43:3B:18:XX handle=33 Status=0
```

The communication stays there without any further bluetooth message. 

Here's another case where I would receive different errors and my new android just wouldn't let me connect. The first few lines are errors I have accumulated in an array and displaying on button press to help troubleshoot.
```
08-01 10:57:24.211  3871  3988 I ReactNativeJS: [ { [Error: Disconnected]
08-01 10:57:24.211  3871  3988 I ReactNativeJS:     line: 193324,
08-01 10:57:24.211  3871  3988 I ReactNativeJS:     column: 22,
08-01 10:57:24.211  3871  3988 I ReactNativeJS:     sourceURL: 'http://192.168.1.30:8081/index.delta?platform=android&dev=true&minify=false' },
08-01 10:57:24.211  3871  3988 I ReactNativeJS:   { [Error: Device 00:13:43:3B:18:6C was disconnected]
08-01 10:57:24.211  3871  3988 I ReactNativeJS:     errorCode: 201,
08-01 10:57:24.211  3871  3988 I ReactNativeJS:     attErrorCode: 8,
08-01 10:57:24.211  3871  3988 I ReactNativeJS:     iosErrorCode: null,
08-01 10:57:24.211  3871  3988 I ReactNativeJS:     androidErrorCode: null,
08-01 10:57:24.211  3871  3988 I ReactNativeJS:     reason: 'Disconnected from 00:13:43:3B:18:6C with status 8 (GATT_INSUF_AUTHORIZATION)',
08-01 10:57:24.211  3871  3988 I ReactNativeJS:     line: 193768,
08-01 10:57:24.211  3871  3988 I ReactNativeJS:     column: 72,
08-01 10:57:24.211  3871  3988 I ReactNativeJS:     sourceURL: 'http://192.168.1.30:8081/index.delta?platform=android&dev=true&minify=false' },
08-01 10:57:24.211  3871  3988 I ReactNativeJS:   { [Error: Device 00:13:43:3B:18:6C was disconnected]
08-01 10:57:24.211  3871  3988 I ReactNativeJS:     errorCode: 201,
08-01 10:57:24.211  3871  3988 I ReactNativeJS:     attErrorCode: null,
08-01 10:57:24.211  3871  3988 I ReactNativeJS:     iosErrorCode: null,
08-01 10:57:24.211  3871  3988 I ReactNativeJS:     androidErrorCode: 133,
08-01 10:57:24.211  3871  3988 I ReactNativeJS:     reason: 'Disconnected from 00:13:43:3B:18:6C with status 133 (GATT_ERROR)',
08-01 10:57:24.211  3871  3988 I ReactNativeJS:     line: 193768,
08-01 10:57:24.211  3871  3988 I ReactNativeJS:     column: 72,
08-01 10:57:24.211  3871  3988 I ReactNativeJS:     sourceURL: 'http://192.168.1.30:8081/index.delta?platform=android&dev=true&minify=false' } ]
08-01 10:57:34.504  3871  6120 D BluetoothGatt: connect() - device: 00:13:43:3B:18:XX, auto: false
08-01 10:57:34.505  3871  6120 D BluetoothGatt: registerApp()
08-01 10:57:34.506  3871  6120 D BluetoothGatt: registerApp() - UUID=c4899d1c-8cb6-4423-9fe9-658f72e07f4d
08-01 10:57:34.610  3871  3888 D BluetoothGatt: onClientRegistered() - status=0 clientIf=7
08-01 10:57:34.780  1472  1862 E bt_btif : bta_gattc_cache_load: can't open GATT cache file /data/misc/bluetooth/gatt_cache_0013433b186c for reading, error: No such file or directory
08-01 10:57:34.815  1472  1571 E BluetoothRemoteDevices: aclStateChangeCallback: State:Connected to Device:00:13:43:3B:18:XX, linktype is 2
08-01 10:57:34.827  3871  3888 D BluetoothGatt: onClientConnectionState() - status=0 clientIf=7 device=00:13:43:3B:18:XX
08-01 10:57:34.862  3871  6120 D BluetoothGatt: discoverServices() - device: 00:13:43:3B:18:XX
08-01 10:57:35.414  3871  3889 D BluetoothGatt: onClientConnParamsChanged() - Device=00:13:43:3B:18:XX interval=6 status=0
08-01 10:57:35.858  3871  3888 D BluetoothGatt: onSearchComplete() = Device=00:13:43:3B:18:XX Status=0
08-01 10:57:35.875  3871  3989 D BluetoothGatt: setCharacteristicNotification() - uuid: 18cda784-4bd3-4370-85bb-bfed91ec86af enable: true
08-01 10:57:35.928  3871  3888 D BluetoothGatt: onClientConnParamsChanged() - Device=00:13:43:3B:18:XX interval=39 status=0
08-01 10:57:35.950  3871  3889 W BluetoothGatt: onCharacteristicRead() - Device=00:13:43:3B:18:XX handle=22 Status=0
08-01 10:57:36.047  3871  3888 W BluetoothGatt: onCharacteristicRead() - Device=00:13:43:3B:18:XX handle=33 Status=0
08-01 10:57:38.104  1049  1599 E Watchdog: !@Sync 7 [2018-08-01 10:57:38.103]
08-01 10:58:08.110  1049  1599 E Watchdog: !@Sync 8 [2018-08-01 10:58:08.110]
08-01 10:58:14.404  2182  2287 E ContactsProvider_EventLog: Flush buffer to file cnt : 1 size : 0Kb duration : 1ms lastUpdatedAfter : 56897 ms mFlush_time_threasold : 2000 mCurrentSize : 257
08-01 10:58:38.117  1049  1599 E Watchdog: !@Sync 9 [2018-08-01 10:58:38.116]
08-01 10:59:08.123  1049  1599 E Watchdog: !@Sync 10 [2018-08-01 10:59:08.123]
08-01 10:59:14.507  2182  2287 E ContactsProvider_EventLog: Flush buffer to file cnt : 1 size : 0Kb duration : 1ms lastUpdatedAfter : 60103 ms mFlush_time_threasold : 2000 mCurrentSize : 257
08-01 10:59:38.129  1049  1599 E Watchdog: !@Sync 11 [2018-08-01 10:59:38.129]
08-01 10:59:47.001  3871  6120 D BluetoothGatt: cancelOpen() - device: 00:13:43:3B:18:XX
08-01 10:59:47.007  3871  3889 D BluetoothGatt: onClientConnectionState() - status=0 clientIf=7 device=00:13:43:3B:18:XX
08-01 10:59:47.011  3871  3889 D BluetoothGatt: setCharacteristicNotification() - uuid: 18cda784-4bd3-4370-85bb-bfed91ec86af enable: false
08-01 10:59:47.012  1472  6753 E BtGatt.GattService: registerForNotification() - No connection for 00:13:43:3B:18:XX...
08-01 10:59:47.017  3871  6120 D BluetoothGatt: close()
08-01 10:59:47.017  3871  6120 D BluetoothGatt: unregisterApp() - mClientIf=7
08-01 10:59:48.024  1472  1571 E BluetoothRemoteDevices: aclStateChangeCallback: State:DisConnected to Device:00:13:43:3B:18:XX
08-01 10:59:48.029  6877  6877 E BluetoothEventManager: ACTION_ACL_DISCONNECTED
08-01 10:59:48.031  1483  1692 E BluetoothEventManager: ACTION_ACL_DISCONNECTED
08-01 10:59:48.049  1472  1571 E BluetoothRemoteDevices: devicePropertyChangedCallback: bdDevice: 00:13:43:3B:18:XX, value is empty for type: 241
08-01 10:59:57.068  2256  2256 E SearchServiceStarter: Task 174 failed or timed out. Client 9923526064305787 disconnecting from SearchService!
```

We have an app currently available on the store which was working fine for me with a samsung SM-T113 (samsung tab e-lite with android 4.4.4) that doesn't present as much trouble with this new android. When I restart my device I try to connect to and use the SM-T113, it connects to it without much problem, but even a fresh reboot of my external device doesn't allow the SM-T380 to connect to it.

[... cut ...] e no way to know if it's the library's fault (through the calls of the RxBleAndroid lib) or the device itself.

I actually searched on Google for some hint but the best hint I was able to find (https://stackoverflow.com/questions/23076641/ble-device-bonding-remove-automatically-in-android) is that android has a problem that puts the bonding in a weird state but I have not seen the messages the user was mentionning in my logcat extract.

Any help troubleshooting this would be greatly appreciated.

Cierpliwy, 2018-10-05:
@darkrift You can always download and send HCI logs from your Android device. It's easier to see what's going on in the low-level implementation of Bluetooth stack.

GroupXTech, 2018-10-25:
I'm experiencing the same the same connection errors tested. samsung s7, s8, samsung 10.1
as well when it does connect it is taking anywhere from 20-60 seconds before It finds my device and attempts the connect.
I couldn't find the btsnoop_log.text file  I'm hoping you can extract what you need from the manager dumsys log.

It seems like this 133 is a known issue in the android community. I too hope you have some luck finding a solution.
One possible solution was to delay the connect method by 500 milliseconds to allow the manager to stop scanning.  No luck for me.

[bluetooth_manager_dumpsys.txt](https://github.com/Polidea/react-native-ble-plx/files/2515559/bluetooth_manager_dumpsys.txt)

closed 2019-03-23 by stale[bot]

handsameliu, 2021-08-31:
@GroupXTech thanks