dotintent/react-native-ble-plx #316: react_native_ble_plx_swift.modulemap' not found

hamishwebb, 2018-08-09:
I keep getting this error when building in xcode:

<unknown>:0: error: module map file '/Users/.../Library/Developer/Xcode/DerivedData/...-ddineicerhcwzmemgxqivvkqammg/Build/Products/Debug-iphonesimulator/react-native-ble-plx-swift/react_native_ble_plx_swift.modulemap' not found

as well as:

<unknown>:0: error: failed to emit precompiled header '/Users/.../Library/Developer/Xcode/DerivedData/...-ddineicerhcwzmemgxqivvkqammg/Build/Intermediates.noindex/PrecompiledHeaders/...-Bridging-Header-swift_1YY2538E66OCR-clang_11Y2ANE0GION0.pch' for bridging header '/Users/.../.../.../ios/...-Bridging-Header.h'

I'm new to working with xcode, which may be the problem, I come from a javascript background.

Avinash-Setty, 2018-08-18:
Even I am facing the same issue. Any help will be appreciated.

hamishwebb, 2018-08-20:
By using a pure React Native build everything worked fine, the above was when using Expo.

Cierpliwy, 2018-10-05:
Closing as the author solved the issue.

closed 2018-10-05 by Cierpliwy

TrebuhD, 2018-11-22:
My xcode build works fine, but I get the above error when building with fastlane.

```
<unknown>:0: error: module map file '/Users/../Library/Developer/Xcode/DerivedData/NotaNota-hcfeypxdnfwftnexobelztcqblci/Build/Intermediates.noindex/ArchiveIntermediates/NotaNota (staging)/BuildProductsPath/Release-iphoneos/react-native-ble-plx-swift/react_native_ble_plx_swift.modulemap' not found
<unknown>:0: error: module map file '/Users/../Library/Developer/Xcode/DerivedData/NotaNota-hcfeypxdnfwftnexobelztcqblci/Build/Intermediates.noindex/ArchiveIntermediates/NotaNota (staging)/BuildProductsPath/Release-iphoneos/react-native-ble-plx-swift/react_native_ble_plx_swift.modulemap' not found
```

jogboms, 2019-07-22:
@TrebuhD did you by any chance manage to solve your issue?

smarchante1, 2019-07-22:
Currently having issues with this very thing. Would be nice to know the solution OP found!

ustincameron, 2019-08-02:
Why is this closed? This is happening without Expo.

mi-mazouz, 2019-10-11:
still having this issue!

echezbro, 2019-10-17:
?

nandorojo, 2019-11-07:
I'm having this issue with a bare workflow Expo project. It's running fine when doing `react-native run-ios`, but if I try to run from xCode, I get these errors.

LondonX, 2020-02-24:
open `.xcworkspace` insted of `.xcodeproj` works for me.

Ailton265, 2020-03-09:
I have the same issue. any one found workround?