dotintent/react-native-ble-plx #319: dyld: Library not loaded: @rpath/libswiftCore.dylib after updating to xcode 9.4 from 9.2

Mahaswami, 2018-08-13:
react-native-ble-plx  - 0.9.1
mac-os - 10.13.12
xcode - 9.4
react - 16.3.1
react-native - 0.55

My app works well in Xcode 9.2 using above configurations. 

After updating to xcode 9.4 my app crashes as soon as it launches with the below error in xcode debugger.

dyld: Library not loaded: @rpath/libswiftCore.dylib
  Referenced from: /Users/xxx/Library/Developer/CoreSimulator/Devices/xxxxxx/data/Containers/Bundle/Application/xxxx/xxx.app/Frameworks/BleClientManager.framework/BleClientManager
  Reason: image not found

Cierpliwy, 2018-10-05:
Closing as frameworks are not used anymore starting from version 0.10.

closed 2018-10-05 by Cierpliwy

harshjoshi096, 2019-05-02:
Hello @Cierpliwy  I have a same issue but in my case I'm using version -  "react-native-ble-plx": "^0.8.0",
Can you please help me on priority to resolve this issue. 
Thanks

dariuszseweryn, 2019-05-02:
Try updating to the newest version of the lib. I am using it in `XCode 10.2.1` without issues

harshjoshi096, 2019-05-02:
Hello @dariuszseweryn Thanks for the response,
Actually I am getting errors around 200 after updating to latest version of ble plx, my OS version is 10.14 and xcode version is 10.1
Any Idea regarding this?
Attached is the screenshot.
Please help if you can
[image]

dariuszseweryn, 2019-05-02:
Unfortunately I do not have enough knowledge on different RN/lib/XCode/Swift version interactions to help you. I just know it is possible to make it work

harshjoshi096, 2019-05-03:
@dariuszseweryn  can you please let me know which version of ble-plxshould I use which should be compatible with my OS version 10.14 and xcode version 10.1

dariuszseweryn, 2019-05-03:
Dunno, newest perhaps