dotintent/react-native-ble-plx #320: Services discovery failed

onuroo, 2018-08-15:
Called discoverAllServicesAndCharacteristics gives error BleError: Services discovery failed for device D4:36:39:CC:F3:5E, after my phone connected to thermal printer.

First scanning gives device object following:

[image]

After connection gives device object following:

[image]

In my opinion, problem is null serviceUUIDs.

How can solve this problem ?

Thanks.

axlider, 2018-08-17:
`discoverAllServicesAndCharacteristics`

results in this object:

Device {isConnectable: null, rssi: null, serviceData: null, overflowServiceUUIDs: null, mtu: 23, …}
_manager:BleManager {_eventEmitter: NativeEventEmitter, _uniqueId: 3, _activePromises: Object, …}
id:"66:02:CA:42:D3:F4"
isConnectable:null
localName:null
manufacturerData:null
mtu:23
name:"BlueTooth Printer"
overflowServiceUUIDs:null
rssi:null
serviceData:null
serviceUUIDs:null
solicitedServiceUUIDs:null
txPowerLevel:null

How is it possible to write to the device?

axlider, 2018-08-24:
Here comes Android log messages:

08-24 11:53:53.591 31227 31251 D BtGatt.GattService: onScanFilterParamsConfigured() - clientIf=5, status=0, action=1, availableSpace=16
08-24 11:53:53.591 31227 31251 D BtGatt.ScanManager: callback done for clientIf - 5 status - 0
08-24 11:53:53.591 31227 31268 D BtGatt.ScanManager: configureRegularScanParams() - queue emtpy, scan stopped
08-24 11:53:53.612  8363  8379 I BeaconManager: BleScanHelper.startBleScan - Start filter Scan(SCREEN_OFF)
08-24 11:53:53.613  8363  8379 D BluetoothLeScanner: Start Scan
08-24 11:53:53.613  8363  8379 D BluetoothAdapter: STATE_ON
08-24 11:53:53.614  8363  8379 D BluetoothAdapter: STATE_ON
08-24 11:53:53.614  8363  8379 D BluetoothAdapter: STATE_ON
08-24 11:53:53.615  8363  8379 D BluetoothAdapter: STATE_ON
08-24 11:53:53.616 31227 31322 D BtGatt.GattService: registerClient() - UUID=009bf638-b41e-49cd-b4b8-b11fe7548010
08-24 11:53:53.717 31227 31251 D BtGatt.GattService: onClientRegistered() - UUID=009bf638-b41e-49cd-b4b8-b11fe7548010, clientIf=5, status=0
08-24 11:53:53.718  8363 25729 D BluetoothLeScanner: onClientRegistered() - status=0 clientIf=5 mClientIf=0
08-24 11:53:53.721 31227 31323 D BtGatt.GattService: getScanSettings() Packgage: com.samsung.android.beaconmanager, isForeground: false, scanMode: 100
08-24 11:53:53.725 31227 31269 E BtGatt.GattService: [GSIM LOG]: gsimLogHandler: android.uid.bcmgr, msg: MESSAGE_START_SCAN
08-24 11:53:53.726 31227 31268 D BtGatt.ScanManager: handling starting scan
08-24 11:53:53.729 31227 31268 D BluetoothAdapter: STATE_ON
08-24 11:53:53.730 31227 31268 D BluetoothAdapter: STATE_ON
08-24 11:53:53.731 31227 31268 D BluetoothAdapter: STATE_ON
08-24 11:53:53.732 31227 31268 D BluetoothAdapter: STATE_ON
08-24 11:53:53.734 31227 31268 D BluetoothAdapter: STATE_ON
08-24 11:53:53.735 31227 31268 D BluetoothAdapter: STATE_ON
08-24 11:53:53.737 31227 31251 D BtGatt.GattService: onScanFilterEnableDisabled() - clientIf=5, status=0, action=1
08-24 11:53:53.737 31227 31251 D BtGatt.ScanManager: callback done for clientIf - 5 status - 0
08-24 11:53:53.738 31227 31268 D BtGatt.ScanManager: set filter index= 12 for clientIf= 5
08-24 11:53:53.738 31227 31268 D BtGatt.ScanManager: addFilterToController: 5
08-24 11:53:53.740 31227 31251 D BtGatt.GattService: onScanFilterConfig() - clientIf=5, action = 0 status = 0, filterType=5, availableSpace=31
08-24 11:53:53.740 31227 31251 D BtGatt.ScanManager: callback done for clientIf - 5 status - 0
08-24 11:53:53.740 31227 31268 D BtGatt.ScanManager: configureFilterParamter() : onFoundTimeout | onLostTimeout | onFoundCount | numOfTrackingEntries | scanFilterRssiThreshold | trackableScanFilterRssiThreshold | matchMode
08-24 11:53:53.740 31227 31268 D BtGatt.ScanManager: configureFilterParamter() : __________500__|________10000__|___________1__|___________________0__|___________________-128__|____________________________-128__|________1_
08-24 11:53:53.742 31227 31251 D BtGatt.GattService: onScanFilterParamsConfigured() - clientIf=5, status=0, action=0, availableSpace=15
08-24 11:53:53.742 31227 31251 D BtGatt.ScanManager: callback done for clientIf - 5 status - 0
08-24 11:53:53.742 31227 31268 D BtGatt.ScanManager: set filter index= 3 for clientIf= 5
08-24 11:53:53.742 31227 31268 D BtGatt.ScanManager: addFilterToController: 5
08-24 11:53:53.744 31227 31251 D BtGatt.GattService: onScanFilterConfig() - clientIf=5, action = 0 status = 0, filterType=5, availableSpace=30
08-24 11:53:53.744 31227 31251 D BtGatt.ScanManager: callback done for clientIf - 5 status - 0
08-24 11:53:53.744 31227 31268 D BtGatt.ScanManager: configureFilterParamter() : onFoundTimeout | onLostTimeout | onFoundCount | numOfTrackingEntries | scanFilterRssiThreshold | trackableScanFilterRssiThreshold | matchMode
08-24 11:53:53.744 31227 31268 D BtGatt.ScanManager: configureFilterParamter() : __________500__|________10000__|___________1__|___________________0__|___________________-128__|____________________________-128__|________1_
08-24 11:53:53.746 31227 31251 D BtGatt.GattService: onScanFilterParamsConfigured() - clientIf=5, status=0, action=0, availableSpace=14
08-24 11:53:53.746 31227 31251 D BtGatt.ScanManager: callback done for clientIf - 5 status - 0
08-24 11:53:53.747 31227 31268 E BtGatt.ScanManager: default value of curScanSetting 0 is choosen
08-24 11:53:53.747 31227 31268 D BtGatt.ScanManager: configureRegularScanParams() [LCD OFF] scan param set to 88 / 4992 (units)
08-24 11:53:53.747 31227 31251 D BtGatt.GattService: onScanParamSetupCompleted() status=0, clientIf=5
08-24 11:53:53.748 31227 31251 D BtGatt.GattService: onScanParamSetupCompleted : 0

Cierpliwy, 2018-10-05:
`serviceUUIDs` field is part of advertisement data. I didn't change the API to don't break user's code. 
To actually get services you should:
* Connect to the device `connectToDevice`
* Discover all services and characteristics `discoverAllServicesAndCharacteristics`
* Get available list of services `servicesForDevice`

closed 2018-10-05 by Cierpliwy