dotintent/react-native-ble-plx #321: isWritable fields false

onuroo, 2018-08-15:
Services function returns isWritableWithResponse and isWritableWithoutResponse is false

Therefore can not write anything with writeCharacteristicWithResponseForDevice.
writeCharacteristicWithResponseForDevice gives error " Operation was rejected "

[image]

How I can change this fields and write something ? Thanks.

darkrift, 2018-08-15:
From what I can find, you are trying to write in a characteristic that represents the "Device name" which is fair to be not writable.

onuroo, 2018-08-15:
Yes you are right. But I wonder why this fields return false ?

darkrift, 2018-08-15:
Because this is a field that cannot be written to in order to update it but only be read from, in other words readonly.

Cierpliwy, 2018-10-05:
I think that this was answered by @darkrift clearly. Thanks!

closed 2018-10-05 by Cierpliwy