dotintent/react-native-ble-plx #323: How can I subscribe to activate the sound?

NdozHH, 2018-08-19:
Hello, if I have this feature that corresponds to my device alarm (TrackR Bravo), how can I subscribe to activate the sound?

[image]

NdozHH, 2018-08-20:
[image]

The device has this service, when used with the nrf connect application, the sound is triggered, but I don't know how to do it using RN ble plx. I'm using this code but nothing happens

`ring = () => {
     this.manager.writeCharacteristicWithoutResponseForDevice(this.state.mydevice.id,'1802','2a06','MDI=').then((res) => {
        console.dir("res",res);
      }).catch((error) => {
        console.dir(error)
      })
}`

and this is the log output
[image]

Cierpliwy, 2018-10-05:
Please ask about library usage on [Gitter](https://gitter.im/RxBLELibraries/react-native-ble) channel.

closed 2018-10-05 by Cierpliwy