dotintent/react-native-ble-plx #324: TypeError: undefined is not an object (evaluating '_BleModule.BleModule.createClient')

SergeyMyssak, 2018-08-19:
I'm new to this library. I've done all steps written in this site (https://github.com/Polidea/react-native-ble-plx) successfully. I get this error all the time: TypeError: undefined is not an object (evaluating '_BleModule.BleModule.createClient'). Here is my project at github: https://github.com/SergeyMyssak/BluetoothInit

How to solve this problem?

Esxiel, 2018-08-20:
I'm also in the same rut, I detached my project from Expo, and I did all the steps written https://github.com/Polidea/react-native-ble-plx#android, including the react-native-link that people keep missing.

AlexKotov, 2018-09-12:
Guys, using `react-native link` doesn't add `new BlePackage()` to `MainApplication.java`. Please, fix this.

Cierpliwy, 2018-10-05:
I tested `react-native link` both on normal RN project as well as on detached Expo. It works fine. Please reopen if there are any other problems around this issue.

closed 2018-10-05 by Cierpliwy

shinstev, 2018-11-27:
@Esxiel how did you fix this problem? I get the error when I publish to expo