dotintent/react-native-ble-plx #327: Question: Would this support reading car data from OBD2 Modules

kyle-ssg, 2018-08-23:
This is more of a question so I appreciate if it is out of scope.

I've got a react native app using this module connecting to an OBD2 Vehicle device, do you know if it would be feasible to use this library to display vehicle sensors like standard obd2 apps, e.g. engine rpm?

kyle-ssg, 2018-08-24:
I'm closing this as it definitely is possible by finding the write service/characteristics to write to then sending something like
1. connect to device
2. find the service/characteristic you need to subscribe to in order to send commands
3. send (base64 encoded) AT SP 0

If anyone finds this and wants to learn how, ping me a message.

kyle-ssg, 2018-08-24:
PS Thanks for the great work

closed 2018-08-24 by kyle-ssg

poison8, 2018-09-07:
you can sende AT Commands? haw to send ?

nguyenvanphuc2203, 2021-06-24:
hello  wants to learn how to connect with OBD2 device?

mehthabt, 2022-11-13:
Hello @kyle-ssg 
Do you have any example of what you did to read date from obd devices?

usama-tariq-qbatch, 2024-03-29:
Hey @kyle-ssg I am building a react-native app to communicate with the OBD2 elm237 devices for reading car fault codes and live data e.g RPM, but i am stuck a point to retrieving data from OBD2 device, I am only able to receive conected obd device name. I think i am trying to read details from OBD2 device instead of car. I have seen some android apps from playstore that establish conenction with ECU before further communication with car. Can you help in achieving the in my react-native app or any work around to do the same.