dotintent/react-native-ble-plx #330: Crashlytics carthage error in 0.10.0

rafkhan, 2018-08-28:
I currently get this error at build time in 0.10.0. Something about crashlytics not being available. Any ideas what this might be?

```
ld: warning: directory not found for option '-F/Users/rafy/Developer/Ovandi/falcon-light2/ios/../node_modules/react-native-ble-plx/ios/BleClientManager/Carthage/Build/iOS'
ld: framework not found Crashlytics
clang: error: linker command failed with exit code 1 (use -v to see invocation)
```

[image]

My package.json looks like this:
```
{
  "name": "FalconLight",
  "version": "0.1.0",
  "private": true,
  "devDependencies": {
    "jest-expo": "25.0.0",
    "react-native-scripts": "1.11.1",
    "react-test-renderer": "16.2.0"
  },
  "main": "./node_modules/react-native-scripts/build/bin/crna-entry.js",
  "scripts": {
    "start": "react-native-scripts start",
    "eject": "react-native-scripts eject",
    "android": "react-native-scripts android",
    "ios": "react-native-scripts ios",
    "test": "node node_modules/jest/bin/jest.js"
  },
  "jest": {
    "preset": "jest-expo"
  },
  "dependencies": {
    "@expo/vector-icons": "^6.3.1",
    "analytics-react-native": "^1.2.0",
    "expo": "^27.0.1",
    "prop-types": "^15.6.2",
    "react": "16.3.1",
    "react-native": "https://github.com/expo/react-native/archive/sdk-27.0.0.tar.gz",
    "react-native-ble-plx": "^0.10.0",
    "react-native-elements": "^0.19.1",
    "react-native-slider": "^0.11.0",
    "react-native-super-grid": "^2.4.3",
    "react-native-vector-icons": "^4.6.0",
    "react-navigation": "^1.5.11",
    "react-navigation-redux-helpers": "^1.0.5",
    "react-redux": "^5.0.6",
    "redux": "^3.7.2",
    "redux-observable": "^0.18.0",
    "redux-segment": "^1.6.2",
    "redux-thunk": "^2.2.0",
    "rxjs": "^5.5.10",
    "victory-native": "^0.18.2"
  }
}
```

rafkhan, 2018-08-28:
My Podfile also looks like this

```
source 'https://github.com/CocoaPods/Specs.git'
platform :ios, '9.0'

EXPO_CPP_HEADER_DIR = 'ExpoKit'

target 'falconlight' do
  pod 'react-native-ble-plx', :path => '../node_modules/react-native-ble-plx'
  pod 'react-native-ble-plx-swift', :path => '../node_modules/react-native-ble-plx'
  
  pod 'ExpoKit',
    :git => "http://github.com/expo/expo.git",
    :tag => "ios/2.6.5",
    :subspecs => [
      "Core",
      "CPP",
      "GL"
    ],
    :inhibit_warnings => true

  pod 'React',
    :path => "../node_modules/react-native",
    :inhibit_warnings => true,
    :subspecs => [
      "Core",
      "ART",
      "RCTActionSheet",
      "RCTAnimation",
      "RCTCameraRoll",
      "RCTGeolocation",
      "RCTImage",
      "RCTNetwork",
      "RCTText",
      "RCTVibration",
      "RCTWebSocket",
      "DevSupport",
      "CxxBridge"
    ]
  pod 'yoga',
    :path => "../node_modules/react-native/ReactCommon/yoga",
    :inhibit_warnings => true
  pod 'DoubleConversion',
    :podspec => "../node_modules/react-native/third-party-podspecs/DoubleConversion.podspec",
    :inhibit_warnings => true
  pod 'Folly',
    :podspec => "../node_modules/react-native/third-party-podspecs/Folly.podspec",
    :inhibit_warnings => true
  pod 'glog',
    :podspec => "../node_modules/react-native/third-party-podspecs/glog.podspec",
    :inhibit_warnings => true

  post_install do |installer|
    installer.pods_project.main_group.tab_width = '2';
    installer.pods_project.main_group.indent_width = '2';

    installer.pod_targets.each do |target|

    if target.pod_name == 'ExpoKit'
      target.native_target.build_configurations.each do |config|
        config.build_settings['GCC_PREPROCESSOR_DEFINITIONS'] ||= ['$(inherited)']
        config.build_settings['GCC_PREPROCESSOR_DEFINITIONS'] << 'EX_DETACHED=1'
        
        # needed for GoogleMaps 2.x
        config.build_settings['FRAMEWORK_SEARCH_PATHS'] ||= []
        config.build_settings['FRAMEWORK_SEARCH_PATHS'] << '${PODS_ROOT}/GoogleMaps/Base/Frameworks'
        config.build_settings['FRAMEWORK_SEARCH_PATHS'] << '${PODS_ROOT}/GoogleMaps/Maps/Frameworks'
      end
    end

    if ['Amplitude-iOS','Analytics','AppAuth','Branch','CocoaLumberjack','FBSDKCoreKit','FBSDKLoginKit','FBSDKShareKit','GPUImage','JKBigInteger2'].include? target.pod_name
      target.native_target.build_configurations.each do |config|
        config.build_settings['IPHONEOS_DEPLOYMENT_TARGET'] = '9.0'
      end
    end
    if target.pod_name == "#{EXPO_CPP_HEADER_DIR}"
      target.native_target.build_configurations.each do |config|
        config.build_settings['CLANG_WARN_COMMA'] = false
        config.build_settings['CLANG_WARN_UNGUARDED_AVAILABILITY'] = false
      end
    end
    # Can't specify this in the React podspec because we need
    # to use those podspecs for detached projects which don't reference ExponentCPP.
    if target.pod_name.start_with?('React')
      target.native_target.build_configurations.each do |config|
        config.build_settings['IPHONEOS_DEPLOYMENT_TARGET'] = '9.0'
        config.build_settings['HEADER_SEARCH_PATHS'] ||= ['$(inherited)']
        config.build_settings['HEADER_SEARCH_PATHS'] << "${PODS_ROOT}/Headers/Public/#{EXPO_CPP_HEADER_DIR}"
      end
    end
    # Build React Native with RCT_DEV enabled
    next unless target.pod_name == 'React'
    target.native_target.build_configurations.each do |config|
      config.build_settings['GCC_PREPROCESSOR_DEFINITIONS'] ||= ['$(inherited)']
      config.build_settings['GCC_PREPROCESSOR_DEFINITIONS'] << 'RCT_DEV=1'
    end

    end
  end
end

```

Cierpliwy, 2018-10-05:
It looks like an unrelated problem. Following line:
`ld: warning: directory not found for option '-F/Users/rafy/Developer/Ovandi/falcon-light2/ios/../node_modules/react-native-ble-plx/ios/BleClientManager/Carthage/Build/iOS'
`
just mentions that you didn't properly clean old 0.9.0 library setup before upgrading to 0.10.0. It's just a warning though.

closed 2019-01-04 by dariuszseweryn