dotintent/react-native-ble-plx #336: Configuring React-native with React-native-ble-plx

ibrahimAyyoub, 2018-09-04:
So I was following this guide: "https://blog.expo.io/so-you-want-to-build-a-bluetooth-app-with-react-native-and-expo-6ea6a31a151d"

At "3--IOS Setup" in first step, "pod install" I get this error I've got no idea why, I followed the exact guide.

```
Ibrahims-MacBook-Pro:appTestibrahim$ cd ./ios
Ibrahims-MacBook-Pro:ios ibrahim$ pod install
Analyzing dependencies
Fetching podspec for `DoubleConversion` from `../node_modules/react-native/third-party-podspecs/DoubleConversion.podspec`
Pre-downloading: `ExpoKit` from `http://github.com/expo/expo.git`, tag `ios/2.5.10`
Fetching podspec for `Folly` from `../node_modules/react-native/third-party-podspecs/Folly.podspec`
Fetching podspec for `React` from `../node_modules/react-native`
Fetching podspec for `glog` from `../node_modules/react-native/third-party-podspecs/glog.podspec`
Fetching podspec for `react-native-ble-plx` from `../node_modules/react-native-ble-plx`
Fetching podspec for `yoga` from `../node_modules/react-native/ReactCommon/yoga`
[!] Unable to find a specification for `react-native-ble-plx-swift` depended upon by `react-native-ble-plx`
Ibrahims-MacBook-Pro:ios ibrahim$ 
```

this is the error that I get:

`[!] Unable to find a specification for `react-native-ble-plx-swift` depended upon by "react-native-ble-plx"`

frederikroeper, 2018-09-06:
I have the same issue

bradyrichmond, 2018-09-21:
Same issue here. I don't totally understand cocoapods, but the only place I see a reference to react-`native-ble-plx-swift` is in the `native-ble-plx-swift.podspec.json` file, which seems to be generated every time you run `pod install`? I tried just commenting out the dependency in that file and it gets uncommented on running that command. I don't get where it's seeing this dependency.

bradyrichmond, 2018-09-21:
Found the solution! You just need to add this line to your `./ios/Podfile`:

`pod 'react-native-ble-plx-swift', :path => '../node_modules/react-native-ble-plx'`.

Then run `pod install` again.

closed 2019-01-04 by dariuszseweryn

diegochavez, 2019-11-05:
If you have this error message "`react-native-ble-plx-swift` does not specify a Swift version"

It works after adding 
`ENV['SWIFT_VERSION'] = '3'` 

Thanks, guys 👍

isa3bel, 2019-12-16:
where do you add `EnV[SWIFT-VERSIOn] = 3`?

leechenghsiu, 2019-12-16:
@isa3bel at the top of your Podfile

williamgoodhew, 2020-06-19:
I am getting the following error when I run pod install

"react": "16.11.0",
"react-native": "0.62.2",
"react-native-ble-plx": "^2.0.0",

'[!] No podspec found for `react-native-ble-plx-swift` in `../node_modules/react-native-ble-plx`'

I have added the following 2 lines to the podfile

'pod 'react-native-ble-plx', :path => '../node_modules/react-native-ble-plx'
  pod 'react-native-ble-plx-swift', :path => '../node_modules/react-native-ble-plx''

williamgoodhew, 2020-06-19:
> I am getting the following error when I run pod install
>
> "react": "16.11.0",
> "react-native": "0.62.2",
> "react-native-ble-plx": "^2.0.0",
>
> '[!] No podspec found for `react-native-ble-plx-swift` in [...]

Note: You no longer need the following line with "react-native-ble-plx": "^2.0.0",
pod 'react-native-ble-plx-swift', :path => '../node_modules/react-native-ble-plx''

architect-aimonkey, 2021-03-05:
> I am getting the following error when I run pod install
>
> "react": "16.11.0",
> "react-native": "0.62.2",
> "react-native-ble-plx": "^2.0.0",
>
> '[!] No podspec found for `react-native-ble-plx-swift` in [...]

Same here!!!