dotintent/react-native-ble-plx #337: RxSwift Binary for Xcode10

kchris78, 2018-09-10:
When will be available RxSwift binary useable for Xcode10? At the moment it crashes with the error:
dyld: Symbol not found: __T0BOSgXwWV
  Referenced from: /Users/kkovacs/Library/Developer/CoreSimulator/Devices/2D21788D-8E16-461C-9D4C-DC9D92650BF9/data/Containers/Bundle/Application/C9376551-F934-4563-93F0-3BFDC923F747/Narke.app/Frameworks/RxSwift.framework/RxSwift
  Expected in: /Users/kkovacs/Library/Developer/CoreSimulator/Devices/2D21788D-8E16-461C-9D4C-DC9D92650BF9/data/Containers/Bundle/Application/C9376551-F934-4563-93F0-3BFDC923F747/Narke.app/Frameworks/libswiftCore.dylib
 in /Users/kkovacs/Library/Developer/CoreSimulator/Devices/2D21788D-8E16-461C-9D4C-DC9D92650BF9/data/Containers/Bundle/Application/C9376551-F934-4563-93F0-3BFDC923F747/Narke.app/Frameworks/RxSwift.framework/RxSwift

It would be a great help, as the launch of the final Xcode10 is soon!!!

valpinkman, 2018-09-11:
I'm actually having the same issue right now. As OP just said, it'd be awesome to test out Xcode 10 and iOS 12 before the imminent release

kchris78, 2018-09-18:
Xcode10 final is released. RxSwift 4.3 is released! Could you include this version in BLE-PLX? So this module could be used under Xcode10! (RxSwift 4.3 supports now Xcode10)

latelt-j, 2018-09-18:
I have the same error, I am not sure to understand the issue @kchris78

kchris78, 2018-09-19:
as I can see when I install this module (v0.10) with npm it downloads the binary version of RxSwift (v4.2). This rxswift version does not work with xcode 10. RxSwift v4.3 should work with Xcode10.

Cierpliwy, 2018-09-19:
Xcode10 support is definitely on my priority list. However v0.10 doesn't download any frameworks and I'm little bit confused by your statement @kchris78.

kchris78, 2018-09-19:
Thanks, confused now too :). Trying to reproduce the downloading when installed BLE-PLX.. I see a big Carthage library under BleClientManager, with RxSwift and RxBluetoothKit checkouts and binaries..

kchris78, 2018-09-19:
now trying to update and recompile the whole thing with xcode10.. maybe i'll be lucky and will run...

kchris78, 2018-09-19:
recompiled, the result is:
dyld: Symbol not found: __T014RxBluetoothKit0B5ErrorON
  Referenced from: /Users/kkovacs/Library/Developer/CoreSimulator/Devices/2D21788D-8E16-461C-9D4C-DC9D92650BF9/data/Containers/Bundle/Application/2F97F73D-64DC-4724-B68C-5AA888399870/Narke.app/Frameworks/BleClientManager.framework/BleClientManager
  Expected in: /Users/kkovacs/Library/Developer/CoreSimulator/Devices/2D21788D-8E16-461C-9D4C-DC9D92650BF9/data/Containers/Bundle/Application/2F97F73D-64DC-4724-B68C-5AA888399870/Narke.app/Frameworks/RxBluetoothKit.framework/RxBluetoothKit
 in /Users/kkovacs/Library/Developer/CoreSimulator/Devices/2D21788D-8E16-461C-9D4C-DC9D92650BF9/data/Containers/Bundle/Application/2F97F73D-64DC-4724-B68C-5AA888399870/Narke.app/Frameworks/BleClientManager.framework/BleClientManager
(lldb) 

Dunno what to do with this, maybe the code in BLE Manager should be updated as you wrote @Cierpliwy . Thank you for your help.

wurmr, 2018-09-21:
Sorry for the ignorant question, @kchris78 says the error is resolved, but that seems like a behind the scene fix.

Does what you did provide a clue to how to fix this package in the next version?

poptocrack, 2018-09-21:
After a call with Polidea, they said this should be fixed next week. No real date yet, but @Cierpliwy will get in touch asap, most likely here.

kchris78, 2018-09-24:
What I think react-native-plx would need is to upgrade the RxBluetoothKit header (or what it needs) files from specifix 4.0.3 to 5.x. After I upgraded the RxBluetoothKit to 5.x I could not compile react-native-plx - I had to downgrade.

Cierpliwy, 2018-09-25:
Today I tested library setup with XCode10 for [pure react native](https://github.com/Cierpliwy/SensorTag) and [expo based](https://github.com/Cierpliwy/SensorTagExpo) projects. I added links to the README file as well. I didn't find any issues with compilation steps and archiving. Could you test above projects if they are working for you?

@kchris78 React native 0.10 is not using frameworks. Make sure you cleaned up your project settings from any leftovers from previous versions of the library.

poptocrack, 2018-09-25:
Can this issue be related with the use of another version than the 0.10? We are using the 0.9+ at the moment.

Cierpliwy, 2018-09-25:
Yes. I strongly suggest to update to 0.10 as there were no API breaking changes and build process is more straightforward. It's really hard to support frameworks in RN projects.

poptocrack, 2018-09-25:
Ok, I will test this asap, probably at the end of the week.

poptocrack, 2018-09-28:
After some tests, all seems to be fine with the 0.10 version. Thanks for your time !

Cierpliwy, 2018-10-05:
Everything should work fine both for version 0.10 and 1.0.0. Please reopen if there are any remaining issues.

closed 2018-10-05 by Cierpliwy