dotintent/react-native-ble-plx #339: Unable to understand or decode characteristic value, after monitoring it

daniyal-intellicel, 2018-09-18:
I am trying to connect and get a Bluetooth thermometer value, here is the device

[Device](https://www.pyleaudio.com/sku/PHTM60BTBL/Bluetooth-Non-Contact-Infrared-Handheld-Thermometer-with-Digital-LCD-Display-Readout)

I am connecting the device, then -> discoverAllServicesAndCharacteristics, and then -> monitoring the characteristic, 
The button press returning me a new characteristic with new value every time, but I am unable to understand or decode the value format.
value1= BFMNAP4J
value2= BE8NAP4J
value3= BFwNAP4J
value4= BFcNAP4J

[image]

Tilwin90, 2018-09-19:
The received values are encoded as base64 strings.

1) For React Native applications you can install the base-64 node package as I don't know of any any out-of-the-box decoding functions (https://www.npmjs.com/package/base-64)

Then your callback could look something like this:
```
... (err, chr) => {
 if (chr && chr.value) {
   const decodedValue = base64.decode(chr.value);
   const value = parseFloat(decodedValue);
   // do something with value here
 }
}
```

daniyal-intellicel, 2018-09-19:
Thanks @Tilwin90 
Although I have tried to decode it with base64-js already but also tried base-64, here are the results.
[image]

karlrl, 2018-09-27:
@Tilwin90's approach is on the right track. If you're using `base-64` @daniyal-intellicel you can get a byte array with `base64.toByteArray()`. Then, you can use a [DataView](https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Global_Objects/DataView) to extract whatever values you sent along. E.g. if you're sending a single, 32-bit float:

```js
const bytes = base64.toByteArray(characteristic.value);
const view = new DataView(bytes.buffer);
const value = view.getFloat32()
```

I'm not sure if that's exactly what the thermometer is sending, but you can adjust what type of value you're extracting, at what offset and the endianness with the `DataView` accessors (e.g. see docs for [`DataView.getFloat32()`](https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Global_Objects/DataView/getFloat32)).

Cierpliwy, 2018-10-05:
value1= BFMNAP4J
value2= BE8NAP4J
value3= BFwNAP4J
value4= BFcNAP4J

value1= 04530d00fe09
value2= 044f0d00fe09
value3= 045c0d00fe09
value4= 04570d00fe09

value1[1] = 83
value2[1] = 79
value3[1] = 92
value4[1]= 87

I guess that first byte is a length of a data. Only second byte changes, which probably means that it contains temperature value. It might be a fixed point value or some value which should be multiplied by a constant (for example 1.1). To be sure you should check the proper documentation.

closed 2019-01-04 by dariuszseweryn

dyarfaradj, 2019-12-08:
> @Tilwin90's approach is on the right track. If you're using `base-64` @daniyal-intellicel you can get a byte array with `base64.toByteArray()`. Then, you can use a [DataView](https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Global_Objects/DataView) to extract whatever values you sent along. E.g. if you're receiving a single, 32-bit float:
> 
> ```js
> const bytes = base64.toByteArray(characteristic.value);
> const view = new DataView(bytes.buffer);
> const value = view.getFloat32()
> ```
> 
> I'm not sure if that's exactly what the thermometer is sending, but you can adjust what type of value you're extracting, at what offset and the endianness with the `DataView` accessors (e.g. see docs for [`DataView.getFloat32()`](https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Global_Objects/DataView/getFloat32)).

In react native, base64 is not supported, other base64 libs for react native doesn't include .tobyteArray

Is there anyway I convert base64 to byte array in react native? Searched without any results, I just want to decode the base64 data sent from the ble so I can read it's value.

dyarfaradj, 2019-12-08:
> value1= BFMNAP4J
> value2= BE8NAP4J
> value3= BFwNAP4J
> value4= BFcNAP4J
>
> value1= 04530d00fe09
> value2= 044f0d00fe09
> value3= 045c0d00fe09
> value4= 04570d00fe09
>
> value1[1] = 83
> value2[1] = 79
> value3[1] = 92 [...]

In react native, base64 is not supported, other base64 libs for react native doesn't include .tobyteArray

Is there anyway I convert base64 to byte array in react native? Searched without any results, I just want to decode the base64 data sent from the ble so I can read it's value.

hatefk, 2021-06-21:
> The received values are encoded as base64 strings.
>
> 1. For React Native applications you can install the base-64 node package as I don't know of any any out-of-the-box decoding functions [...]

Looks like this package does not support toByteArray() anymore:
https://www.npmjs.com/package/base-64

daniel1up, 2021-07-15:
Hello, 
I have the same problem with converting value to a right meaning value. 

```
const bytes = base64js.toByteArray(result.value);
console.log("bytes: ", bytes);
// bytes:  [15, 0, 100, 34, 148, 16, 100]
```
I got this response after using toByteArray, but I don't know what to do to get the right value.
I am connecting to IQOS device to get the batery status.

Does anyone have any solution?