dotintent/react-native-ble-plx #340: [Android] Checking location provider enabled fails

Tilwin90, 2018-09-20:
I want to report a different behavior of the same application ran physically on 
- a Samsung Galaxy S7 (Android Version 8.0, with Bluetooth 4.2.) -> works correctly (both debug and release mode)
- a Samsung Galaxy S8 (Android Version 8.0, with Bluetooth 5.0) -> fails (both debug and release mode)
- a One Plus 5t (Android Version 8.1, with Bluetooth 5.0) -> fails (release mode)

**Context:**
- Regular device discovery using 
`bleManager.startDeviceScan(uuids, {allowDuplicates: false }, (err,device) => callback(err,device))
` 
- On the devices where it fails, it always hits the error "Location services are disabled"

**Configurations:**
- Since all devices are running on Oreo, it's a no-brainer if the configurations work on one of them it should also work on the rest (unless I'm missing something and for bluetooth 5.0 you need some custom mumbo jumbo)
- I have added the following mandatory lines of code in the manifest:
```
<uses-permission android:name="android.permission.BLUETOOTH"/>
<uses-permission android:name="android.permission.BLUETOOTH_ADMIN"/>
<uses-permission-sdk-23 android:name="android.permission.ACCESS_COARSE_LOCATION"/>
<uses-feature android:name="android.hardware.bluetooth_le" android:required="true"/>
```
- We implemented ask for location permission and it works smoothly on all phones - naturally I checked in Settings -> Apps -> [My App] -> Permissions and indeed 'Location' is checked appropriately

**Detailed debugging:**
I naturally dug into the Android code and discovered that the fail happens somewhere deep in your rxandoidble package @CheckerLocationProvider.isLocationProviderEnabled() method, namely at this line:
`return Settings.Secure.getInt(contentResolver, Settings.Secure.LOCATION_MODE) != Settings.Secure.LOCATION_MODE_OFF;`
The method call always returns 0 (equivalent of Settings.Secure.LOCATION_MODE_OFF)!

If there's any other information I can give you about this please let me know because at this point I am stuck.

Tilwin90, 2018-09-26:
Sorry for the false alarm.
It seems that the location services on the device need to be enabled for the Bluetooth to work.

closed 2018-09-26 by Tilwin90

naveenkrishnous, 2019-04-11:
> Sorry for the false alarm.
> It seems that the location services on the device need to be enabled for the Bluetooth to work.

HI Tilwin90, is there any specific reason on why location service should be enabled for Bluetooth? Does enabled Bluetooth wouldn't suffice?

naveenkrishnous, 2019-04-11:
@Tilwin90 is there any specific reason on why location service should be enabled for Bluetooth? Does enabled Bluetooth wouldn't suffice?

> Sorry for the false alarm.
> It seems that the location services on the device need to be enabled for the Bluetooth to work.

@Tilwin90 is there any specific reason on why location service should be enabled for Bluetooth? Does enabled Bluetooth wouldn't suffice?

emech-en, 2019-07-02:
> Sorry for the false alarm.
> It seems that the location services on the device need to be enabled for the Bluetooth to work.

Why location service is required?

naveenkrishnous, 2019-07-03:
> > Sorry for the false alarm.
> > It seems that the location services on the device need to be enabled for the Bluetooth to work.
> 
> Why location service is required?

Thanks. Got it earlier clarified from android developer website.

https://developer.android.com/about/versions/marshmallow/android-6.0-changes.html#behavior-hardware-id