dotintent/react-native-ble-plx #341: [Android] Crash when triggering a scan

tsyirvo, 2018-09-22:
Hello, I’m trying to use this lib to scan BLE devices on Android but when calling `this.manager.startDeviceScan` the app crash with no log whatsoever with the native popup `MyApp has stopped.`.
On iOS I have no problem scanning and finding the device I want to connect to.
Do I need to do something different on Android ?

I set the log level to verbose but nothing is shown and I can successfully use some other methods form the manager without issues.
I tried passing a list of UUIDs or some options to `startDeviceScan` but the app always crash.

I'm using react-native 0.54.4 and react-native-ble-plx 0.10.0

Cierpliwy, 2018-10-04:
I didn't spot such a situation. Did you enable permissions before starting scanning?

renanborgez, 2018-10-05:
@tsyirvo 
Hello, i had the same problem as you, and in my case the problem was the JDK version. I was using the version 10, and it **must** be 8. You can check it

tsyirvo, 2018-10-09:
Hi, downgrading the JDK to version 8 like @renanborgez said seems too work :)
The weird thing is that for some time I manage to make it work with version 10 and at one time it stopped right after I linked a new native module... I downgraded after seeing this and I started working again right after !
If this is truly linked to the JDK version, it should be listed as such on the README to prevent other folks asking this question.

renanborgez, 2018-10-10:
Yeah, i'm agree with @tsyirvo 

The point is, if the version 10 is not supported and it is an expected behaviour for the version 10 or it is actually a bug that no one has catch it yet.

In any case we should have a action on it, or to solve the bug or to document it on README

Cierpliwy, 2018-10-11:
@dariuszseweryn Can it be somehow related to RxAndroidBle?

dariuszseweryn, 2018-10-11:
Android OS does not work with anything that's above Java 8

ujwal-setlur, 2018-10-20:
Phew, thanks so much for having this issue logged. Saved me a lot of time!

closed 2019-01-04 by dariuszseweryn