dotintent/react-native-ble-plx #343: Getting error in Android - java.lang.IllegalStateException: BleManager not created when tried to start device scan

harshjoshi096, 2018-09-28:
Hello
I am getting an error in android but in iOS its working fine. 
The error says - 
`ReactNative: Exception in native call
                                                                             java.lang.IllegalStateException: BleManager not created when tried to start device scan
                                                                                 at com.polidea.reactnativeble.BleModule.safeStartDeviceScan(BleModule.java:273)
                                                                                 at com.polidea.reactnativeble.BleModule.startDeviceScan(BleModule.java:268)
                                                                                 at java.lang.reflect.Method.invoke(Native Method)
                                                                                 at com.facebook.react.bridge.JavaMethodWrapper.invoke(JavaMethodWrapper.java:374)
                                                                                 at com.facebook.react.bridge.JavaModuleWrapper.invoke(JavaModuleWrapper.java:162)
                                                                                 at com.facebook.react.bridge.queue.NativeRunnable.run(Native Method)
                                                                                 at android.os.Handler.handleCallback(Handler.java:754)
                                                                                 at android.os.Handler.dispatchMessage(Handler.java:95)
                                                                                 at com.facebook.react.bridge.queue.MessageQueueThreadHandler.dispatchMessage(MessageQueueThreadHandler.java:31)
                                                                                 at android.os.Looper.loop(Looper.java:163)
                                                                                 at com.facebook.react.bridge.queue.MessageQueueThreadImpl$3.run(MessageQueueThreadImpl.java:194)
                                                                                 at java.lang.Thread.run(Thread.java:760)`

Cierpliwy, 2018-10-04:
You error is suggesting that you didn't create BleManager instance properly.

harshjoshi096, 2018-10-05:
@Cierpliwy Thanks for your response, but I have created the BleManager instance properly, and its working fine in iOS app, in android when we do continuously scanning for the device in one view then its working but when we come from any other view to previous view, I am receiving this error, also I have checked that the blemanager instance is already created.

harshjoshi096, 2018-10-09:
@Cierpliwy yes, I think that may be a reason, I am reviewing it currently, also in an android device when we are doing `startDeviceScan()`, the device is getting freeze until the whole scanning process. and once it gets complete it works fine, but in my case, I am scanning the device in an interval so the functionality is getting the freeze, any solution for that?

Cierpliwy, 2018-10-09:
What do you mean by freeze? Are you talking about JS thread freeze or native thread freeze? Can you send me code snippets how you are calling `startDeviceScan` and which parameters are used?

harshjoshi096, 2018-10-10:
@Cierpliwy For freeze I mean the app UI is not working until the scanning process gets complete, once device added than all the UI works properly. 
Below is my code snippet -

`this._interval = setInterval(() => {
            let bluetoothFlag = store.getState().device.bluetoothFlag;
            if (!bluetoothFlag) {
                if (!screensArr.includes(Actions.currentScene)) {
                    const subscription = this.manager.onStateChange((state) => {
                        if (state === 'PoweredOn') {             
                            if(this.manager){                             
                                    this.manager.destroy();
                                    this.manager = new BleManager();                                 
                                   const subscription = this.manager.onStateChange((state) => {
                                        if (state === 'PoweredOn') {
                                            if(this.manager){
                                                if(Constant.managerAlreadyCreated == true){
                                                    this.manager.destroy();
                                                    this.manager = new BleManager();
                                                    Constant.managerAlreadyCreated = false
                                                }
                                            }
                                            subscription.remove();
                                        }
                                    }, true);
                                }                    
                            subscription.remove();
                        }
                    }, true);
                }
            }
         }, 10000)`

renanborgez, 2018-10-10:
@harshjoshi096 
The freeze happens because the JS UI works using only 1 thread, and the scan function from BleManager locks this thread, in my case i did using settimeout and my UI is not freeze anymore, look bellow:

```js
let _counter = 0;
  Manager.startDeviceScan(null, null, (error, device) => {
    _counter = _counter++;

    // TODO: Proper way to store/return errors
    if (error) {
      this.stopScan();
      return;
    }

    if (device && device.serviceUUIDs) { // I'm just checking if my device has a serviceUUIDS
      listener(device);
    }
   // Here is the magic, counting until 10 cycles is good trick to avoid the freeze
   // and then i stop and startScan again, then you UI ill not freeze
    if (this._counter & 10 === 0) { 
      this.stopScan();
      this.startScan();
    }
  }, true);
```

renanborgez, 2018-10-10:
@harshjoshi096 
The freeze happens because the JS UI works using only 1 thread, and the scan function from BleManager locks this thread, in my case i "solved" this problem using settimeout and my UI is not freeze anymore, look bellow:

PS: You can try to use some library to make this call in another thread, like webworkers and maybe something with Webview

```js
const Manager = new BleManager();

const _deviceScan = listener => {
  let _counter = 0;
  Manager.startDeviceScan(null, null, (error, device) => {
    console.log('found');
    _counter = _counter++;

    // TODO: Proper way to store/return errors
    if (error) {
      this.stopScan();
      return;
    }

    // TODO: Discuss a proper way to check if the device is valid
    if (device && device.serviceUUIDs) {
      listener(device);
    }
    // HERE IS THE MAGIC!!!! it ill free your UI to handle others actions between the stop and startscan
    if (this._counter & 10 === 0) { 
      this.stopScan();
      this.startScan();
    }
  }, true);
};

const _init = listener => {
  const subscription = Manager.onStateChange((state) => {
    if (state === 'PoweredOn') {
      subscription.remove();
      _deviceScan(listener);
    }
  }, true);
};

const stopScan = Manager.stopDeviceScan;
const startScan = listener => {
  _init(listener);
};

const connectToDevice = (device, listener) => {
  Manager.connectToDevice(device.id).then(connected => {
    if (connected) {
      listener(device);
    }
  }).catch((error) => {
    // TODO: Proper way to store errors
    listener(null, error);
  });
};
```

renanborgez, 2018-10-10:
@harshjoshi096 
The freeze happens because the JS UI works using only 1 thread, and the scan function from BleManager locks this thread, in my case i "solved" this problem using a counter and my UI is not freeze anymore, look bellow:

PS: You can try to use some library to make this call in another thread, like webworkers and maybe something with Webview

```js
const Manager = new BleManager();

const _deviceScan = listener => {
  let _counter = 0;
  Manager.startDeviceScan(null, null, (error, device) => {
    console.log('found');
    _counter = _counter++;

    // TODO: Proper way to store/return errors
    if (error) {
      this.stopScan();
      return;
    }

    // TODO: Discuss a proper way to check if the device is valid
    if (device && device.serviceUUIDs) {
      listener(device);
    }
    // HERE IS THE MAGIC!!!! it ill free your UI to handle others actions 
    // because i'm continuously releasing my thread
    if (this._counter & 10 === 0) { 
      this.stopScan();
      this.startScan();
    }
  }, true);
};

const _init = listener => {
  const subscription = Manager.onStateChange((state) => {
    if (state === 'PoweredOn') {
      subscription.remove();
      _deviceScan(listener);
    }
  }, true);
};

const stopScan = Manager.stopDeviceScan;
const startScan = listener => {
  _init(listener);
};

const connectToDevice = (device, listener) => {
  Manager.connectToDevice(device.id).then(connected => {
    if (connected) {
      listener(device);
    }
  }).catch((error) => {
    // TODO: Proper way to store errors
    listener(null, error);
  });
};
```

Cierpliwy, 2018-10-11:
I wouldn't recommend doing something like that:
```js
  if (this._counter & 10 === 0) { 
      this.stopScan();
      this.startScan();
    }
```
On Android restarting scanning too frequently may result in an error and can be blocked for a short period of time by an operating system. 

Listener of `startDeviceScan` shouldn't block JS thread itself. It's probably due to other work which you are executing inside this callback. One option to handle it better is:

* Make sure to do as little amount of work as possible in scanning listener.
* Reduce emission of events by setting `scanMode` to `LowPower` and `allowDuplicates` to `false` (default).

@harshjoshi096 There is no point in destroying and creating `BleManager` all over again. It's an expensive operation.

closed 2019-01-04 by dariuszseweryn

tomdaniel0x01, 2019-11-02:
My React Native UI would also hang while performing a `startDeviceScan`. The cause was having the Debugger enabled. Once I disabled the debugger the UI responded lighting fast while `startDeviceScan` was executing.

Bikash-Sawa, 2021-02-04:
> The error that you encountered, happens only when `rxBleClient` is null. That means:
> 
> * You called `startDeviceScan()` before `new BleManager()` (sets `rxBleClient` to not null)
> * You called `startDeviceScan()` after `bleManager.destory()` (sets `rxBleClient` to null)
> * You are creating multiple managers, which isn't allowed and you have a race condition (one of above cases) due to component lifecycle's methods independence between views. (I assume you are holding BleManager inside them). I strongly recommend creating one instance of `BleManager` outside of the component hierarchy.

I was facing same  issue. We must create BleManager instance outside the  component hierarchy. If we instance inside component then it is creating multiple instances due to which i cannot stop the bleManager scan too.