dotintent/react-native-ble-plx #345: Tutorial needed

calcazar, 2018-10-13:
Howdy!

Some of us are new to mobile development and don't have much experience with bluetooth overall. I was wondering if it was possible to add a tutorial section with perhaps a tile or another bluetooth device that shows how to connect and retrieve/read real data from it.

I have a medical bluetooth device and I can connect and read the services/characteristics without a problem, the issue is that I'm not sure how to decode the data it makes available via the characteristics. I keep getting base64 for the value property which after decoding doesn't make much sense to me: 

Encoded: AZyDslz++CcAGA8AAMA=
Decoded: \'

Any help is much appreciated! I've looked online for tutorials but haven't found one that I was able to get working on my end.

Cierpliwy, 2018-10-13:
Hello :)

Characteristic UUID defines its type. If your UUID is of the following form: XXXXXXXX-0000-1000-8000-00805F9B34FB, that means it's probably defined and available on [Bluetooth site](https://www.bluetooth.com/specifications/gatt/characteristics).

Otherwise, you have to find the specification for your characteristic describing how it should be read.

calcazar, 2018-10-14:
@Cierpliwy I'm getting UUID's that are definitely different from XXXXXXXX-0000-1000-8000-00805F9B34FB

When you say I have to find the specifications, you mean I have to get the specs for the device from the manufacturer correct?

Cierpliwy, 2018-11-23:
Yes, that's right.

closed 2019-01-04 by dariuszseweryn

assafb81, 2019-01-06:
hi @Cierpliwy 
I am struggling with the package :(.
I want to write an app that connect with another device via BLE.
I can't find it in my scan, nor I can't understand what is the `device.id` (It does not look like my IMEI or Bluetooth address).
Currently I am writing my app on iOS but I want to detect iOS and Android devices.
What I need to do beside (1) check if my device (not the one with my app) BT is on and (2) if it is supporting BLE (using BLE checker app)?

thanks for you help