dotintent/react-native-ble-plx #346: Manufacturer-Data always null

dominikeinkemmer, 2018-10-13:
Hi,

I'm currently working on reading the manufacturerData of a connected device because I need to get the firmware-version of my device to give some restrictions inside my app in terms of functionality based on it.

However, if I try to read the **manufacturerData** or **serviceData** I always receive NULL -> when I checked the native code I also realized why, as **_asJSObject_** method of the **Peripheral** class fills these values with NSNull.

I also saw however, that RxBluetoothKit which this library is based on has another class called **ScannedPeripheral** which actually puts that information into its **_asJSObject_** method so I wondered if there is any way to get this information for the **Peripheral** class as well?

Is there currently no way to receive these data with this library?

Thanks in advance.

dominikeinkemmer, 2018-11-07:
@Cierpliwy do you know anything about this maybe? :) You used to reply to most of my issues but this was left hanging for almost a month so I thought I'll ping directly :) sorry if that's not your part of this project :)

Cierpliwy, 2018-11-07:
Manufacturer data field is only available during the scanning process. If it is present during the advertisement, Device instances returned from scanning callback should have this field set. You can cache this information afterward.

Hope it helps :)

dominikeinkemmer, 2018-11-07:
@Cierpliwy thanks for the fast response :) Hmm unfortunately I also get null for these data during the scanning process :/ I will do some research, maybe it has to do with the bluetooth module we use.

thanks 😊

dominikeinkemmer, 2018-11-07:
Thanks for the help, I will reopen if needed yes :)

np-singh, 2023-01-27:
Hey @dominikeinkemmer, A bit late. But were you able to solve this. I am dealing with a similar issue, where manufacturerData I get back is null. 

And thanks @Cierpliwy , I will test and try to cache data during Scan process.