dotintent/react-native-ble-plx #347: [Android] Reconnect when BLE device is out of range (in background)

revolucion, 2018-10-14:
Hi BLE experts,

I know there have been quite some posts on this subject, but I've been struggling with this for weeks and think I've read about everything about it but still can't get my app to function properly.

I've been working on a RN app, which pairs with a peripheral and requires that connection to stay alive. So basically, if (or when) the device goes out of range, I'd need the device to automatically reconnect again.

For iOS this works fine! I've set up the background modes with the Bluetooth Capabilities, and on the disconnect event I simply do another call to ```manager.connectToDevice( <deviceID> )```  and when the device gets in range again, it reconnects. This works when the app is active, and when the app is in background (I guess because the connect method in iOS has no timeout so it keeps looking until it finds it). So happy all together.

But for Android I can *only* get the reconnect to work if the app is open.
Same idea, in the onDisconnected callback I do a call to manager.connectToDevice, and it reconnects fine as long as app is in foreground.
As soon as the screen turns off, or the app is pushed to background, it just won't reconnect until I open the app again. The funny thing is, it still DOES respond to the disconnect event every time, as I display a local notification in the ```onDisconnected``` callback, and that also happens when the app is in background.

So am I doing something horribly wrong? 
I feel like I'm going insane, and even though I have plenty of experience with iOS, I'm pretty new to Android. And I haven't found something similar to Bluetooth Central or a permission I need to set.

The autoConnect flag doesn't seem to make any difference by the way. Or am I supposed to set the autoConnect flag, and not make any calls to manager.connectToDevice or manager.scanForDevices? 

I hope someone can shine some light, I barely have any hair left from pulling it ;)

Ilario17, 2019-03-07:
have you found a solution?

Cierpliwy, 2019-04-05:
Is there anything we can do about it @dariuszseweryn ?

dariuszseweryn, 2019-04-05:
Given [this variety of behaviours](https://dontkillmyapp.com/) I see no clear path that could address all situations.

good-lly, 2019-04-24:
so summary is - we can not use/handle BLE msgs when app is killed (in background) on Android?

how about use of something like [Headless JS](https://github.com/jamesisaac/react-native-background-task)?

AdamGold, 2019-05-06:
Any news on this subject? @dariuszseweryn

dariuszseweryn, 2019-05-06:
Unfortunately no. I have not learned anything that could help

closed 2019-07-12 by stale[bot]

arberhh, 2020-05-08:
@revolucion i know this is an old issue but can u share with how u made the reconnect work on ios.
I'm calling the connect to device on disconnect event, but i haven't been able to reconnect on background

Neel-Narvekar, 2022-06-28:
Any updates since then? @dariuszseweryn

paurosellfarre, 2022-07-22:
Does we have a solution? I think this will make a lot of positive difference  @dariuszseweryn

avanmalleghem, 2024-08-15:
Any update on this one @dariuszseweryn ? Still relevant !