dotintent/react-native-ble-plx #356: Indicating a Characteristic Disconnects the device?

2joocy, 2018-10-30:
Is there a way to indicate for a characteristic, or is it the same as setting up a notification? 

My example is that I want to read a Glucose Measurement, although everytime I do so, it disconnects the device if I try to WriteWithResponse to it.

[image] 

https://www.bluetooth.com/specifications/gatt/viewer?attributeXmlFile=org.bluetooth.service.glucose.xml

My steps:

1. Notifify the Glucose Measurement 

2. Indicate to the Record Access Control Point

3. WriteWithResponse to the Record Access Control Point

Here's my example code:

```
const GLUCOSE_READING = "00002a18"
const RECORD_ACCESS_CONTROL_POINT = "00002a52"

getCharacteristic(device: any, serviceUuid: string, characteristicUuid: string){
		return new Promise((resolve, reject) => {
			device.characteristicsForService(serviceUuid).then((characteristicArr: any) => {
				for(let i = 0; i<characteristicArr.length; i++){
					const uuidSplit = characteristicArr[i].uuid.split("-")
					if(uuidSplit[0] === characteristicUuid){
						resolve(characteristicArr[i])
						break;
					}
				}
			})
		});
		
	}

	setNotification(device: any, characteristic: any){
		return device.monitorCharacteristicForService(characteristic.serviceUUID, characteristic.uuid, (error: any, subChar: any) => {
			console.log(subChar)
		})
	}

	writeToCharacteristic(characteristic: any){
		characteristic.writeWithResponse(Buffer.from("0x0101", 'hex').toString('base64')).then((characteristic : any) => {
			console.log(characteristic)
		})
	}

	scanAndConnect() {
		this.state.bleManager.startDeviceScan([], null, (error: any, scannedDevice: any) => {
			if(scannedDevice && scannedDevice.name){
				console.log(`Found device ${scannedDevice.name}`)
				if(scannedDevice.name.includes("Contour")){
					console.log("Found Contour")
					this.state.bleManager.stopDeviceScan()
					scannedDevice.connect().then((device: any) => {
						this.setState({isScan: !this.state.isScan}, () => {
							device.discoverAllServicesAndCharacteristics().then((servDev:any) => {
								servDev.services().then((servArr: any) => servArr.forEach((service: any) => {
									this.getCharacteristic(servDev, service.uuid, GLUCOSE_READING).then(characteristic => {
										console.log(characteristic)
										const sub = this.setNotification(servDev, characteristic);
										this.getCharacteristic(servDev, service.uuid, RECORD_ACCESS_CONTROL_POINT).then(characteristic => {
											console.log(characteristic)
											this.writeToCharacteristic(characteristic)
										})
									})
								}))
							})
						})
					})
				}
			}
		})
	}
```

Cierpliwy, 2018-11-07:
Is specific supports only indications, calling `monitorCharacteristicForService` will listen for them. I'm not sure why your device disconnects. You can check specific `iosError`, `attError` and `androidError` fields inside Error object. Additionally, I will check logs on the firmware side if you can do that.

Cierpliwy, 2018-11-23:
Closing due to inactivity. Please reopen if needed.

closed 2018-11-23 by Cierpliwy

renatoramossilva, 2020-04-15:
I have the same problem, anyone has the solution for this issue?