dotintent/react-native-ble-plx #359: Read Notification Data, Service

ALAxHxC, 2018-11-05:
Hi people, 
Im trying to get data of pulse oximeter device,
Already connect and identify the device but, when y trying to set service and characteristics with those function:
leer_datos_de_notificacion(device){
    device.monitorCharacteristicForService(
    "cdeacb80-5235-4c07-8846-93a37ee6b86d",
     "cdeacb81-5235-4c07-8846-93a37ee6b86d",
      (error, data) => {
        if (error){
          console.log("Error pidiendo datos",error);
          return;
        }
        console.log("Data de notifiacacion",data)

      },
    ).subscribe((data)=>{
      console.log("Resultado ",data)
    })
  }
The app crash with message:  
undefined its not a funcion.
[image]

Can you helpme?

Cierpliwy, 2018-11-07:
I think you should provide more information:
- which line exactly throws an error
- what are values of objects used in this line.

Cierpliwy, 2018-11-23:
Closing due to inactivity. Please reopen if needed.

closed 2018-11-23 by Cierpliwy

dariuszseweryn, 2019-01-09:
Correct me if I am wrong but `Subscription` has only a method `.remove()`, not `.subscribe()`