dotintent/react-native-ble-plx #361: connectedDevices ocasionally fails to return a connected device

sydorenkos, 2018-11-10:
Before all - thank you so much for such a great library.
I have a series of custom built smart sensors that communicate with android devices. Ocasionally manager.connectedDevices() returns an empty array, even though one sensor is still connected (the sensors have visual feedback of connection status). If I, for example, disable bluetooth on the phone - the sensor responds visually that the connections was lost.
I notice that this happens in a situation when a connection atempt takes more time than normal (I am using a routine that attempts connection several times with ConnectionOptions { timeout: 2000 }, and then the promise finally rejects with an error, however even after the error gets loggeg - the sensor still get connected (one - two seconds after the error was thrown).  Is there another way to check if a device is connected in such situations?

Cierpliwy, 2018-11-23:
Is it happening on iOS? If so please send me logs which are visible on XCode console. You need to call [https://polidea.github.io/react-native-ble-plx/#blemanagersetloglevel](https://polidea.github.io/react-native-ble-plx/#blemanagersetloglevel) to enable that.

sydorenkos, 2018-11-23:
Sorry - forgot to mention - it is happening on Anfdroid. I haven't checked it on iOS.

sydorenkos, 2018-11-23:
Thaks for the idea - I haven't been using logcat. I will run some tests tonight and will post the logs when I 'catch' it.

sydorenkos, 2018-11-27:
After three consecutive days of tests with three different boards  the described problem happended only two times, on two different modules, and both times when the terminal was closed... So far no luck with the usable logcat.  I also notice that it only happened when reception conditions were marginal (sensors are far away). I did get a log from 
the disconnection listener which returned null (supposedly the device was disconected normally), but after the last connection attempt the module, again, showed the connection status as connected.  
In any case I have a couple of questions about possible steps I can take to diagnose this situation more efficiently.
I use widows terminal and logcat tags BluetoothGatt:v BtGatt:v bt-btif:v - am I using the right tags?
My program has a recursive function that attempts 2.5 second connections for five times, before returning a rejected promise. I am wondering if the problem is conected to this, I will definitely look into it. Using this 'hack' did improve connection reliability in situations when reception conditions are not the best. Question - Is it better to use a single call to connectToDevice with a bigger timeout parameter, to make sure the phone can connect (has more time to connect) to distant devices more reliably?