dotintent/react-native-ble-plx #363: Device parameters when connected

kenobi91, 2018-11-18:
Hello, I am having some trouble retrieving device data once the device is connected. When I run startDeviceScan () I console log scanned devices (i only have one) and it has data in it like this:
{
  id: "00:15:85:14:9C:09",
  isConnectable: null,
  localName: "BT05", 
  manufacturerData: "SE0AFYUUnAk=", 
  mtu: 23,
  name: "BT05" ,
  overflowServiceUUIDs: null,
  rssi: -57,
  serviceData: {0000b000-0000-1000-8000-00805f9b34fb: "AAAAAA=="},
  serviceUUIDs: ["0000ffe0-0000-1000-8000-00805f9b34fb"],
  solicitedServiceUUIDs: null,
  txPowerLevel: 0,
  _manager: BleManager-object,
}

after that I do device. connect() and it is successful, but when I console log device I have this data:

{
  id: "00:15:85:14:9C:09",
  isConnectable: null,
  localName: null,
  manufacturerData: null,
  mtu: 23,
  name: "BT05",
  overflowServiceUUIDs: null,
  rssi: null,
  serviceData: null,
  serviceUUIDs: null,
  solicitedServiceUUIDs: null,
  txPowerLevel: null,
  _manager: BleManager-object
}

after that  i tried device.discoverAllServicesAndCharacteristics() but i got same result.
I saw one similar issue and also tried device.services() but i got error:

BleError: Services not discovered for device 00:15:85:14:9C:09
    at parseBleError (blob:http://localhost:8081/132206d4-b879-458d-87b6-d78f1fbb4e95:120653:18)
    at BleManager._callPromise$ (blob:http://localhost:8081/132206d4-b879-458d-87b6-d78f1fbb4e95:121001:51)
    at tryCatch (blob:http://localhost:8081/132206d4-b879-458d-87b6-d78f1fbb4e95:20190:19)
    at Generator.invoke [as _invoke] (blob:http://localhost:8081/132206d4-b879-458d-87b6-d78f1fbb4e95:20365:24)
    at Generator.prototype.(anonymous function) [as throw] (blob:http://localhost:8081/132206d4-b879-458d-87b6-d78f1fbb4e95:20233:23)
    at tryCatch (blob:http://localhost:8081/132206d4-b879-458d-87b6-d78f1fbb4e95:20190:19)
    at invoke (blob:http://localhost:8081/132206d4-b879-458d-87b6-d78f1fbb4e95:20266:22)
    at blob:http://localhost:8081/132206d4-b879-458d-87b6-d78f1fbb4e95:20278:15
    at tryCallOne (blob:http://localhost:8081/132206d4-b879-458d-87b6-d78f1fbb4e95:21463:14)
    at blob:http://localhost:8081/132206d4-b879-458d-87b6-d78f1fbb4e95:21564:17

Cierpliwy, 2018-11-23:
```
localName: "BT05",
manufacturerData: "SE0AFYUUnAk=",
overflowServiceUUIDs: null,
rssi: -57,
serviceData: {0000b000-0000-1000-8000-00805f9b34fb: "AAAAAA=="},
serviceUUIDs: ["0000ffe0-0000-1000-8000-00805f9b34fb"],
solicitedServiceUUIDs: null,
txPowerLevel: 0,
``
are only visible during scanning. Please call ` device.discoverAllServicesAndCharacteristics()` and after it successfully completes call `device.services()`. It should work.

kenobi91, 2018-11-29:
Thx i already did that and it worked :)