dotintent/react-native-ble-plx #365: iOS 11.1.x specific connection issues.

kevinmeyvaert, 2018-11-19:
Hello there! We have been using react-native-ble-plx for quite some time now, we recently upgraded from 0.10.0 to 1.0.0, but ever since we're having issues connecting to our bluetooth peripheral with iOS devices on iOS 11.1 & iOS 11.1.x

Further testing has shown that connecting with iOS 10.x and iOS 12.x is still working as before without any issues. As we're currently still investigating what the actual issue is, I wanted to reach out to anyone experiencing any issues on iOS 11.1 & iOS 11.1.x

I will update this ticket as we gather more info.

-> downgrading to 0.10.0 fixed the issue for now.

kevinmeyvaert, 2018-11-20:
We found out that the issue is related to the thread @bitcrumb mentioned above. And is a specific bug with iOS versions 11.x

> When the restored peripheral's state is 'connected', canSendWriteWithoutResponse always returns false.

We will be making a pull request to resolve this shortly.

Cierpliwy, 2018-11-23:
Workaround was merged and released in 1.0.1. Please check if it works for you.

bitcrumb, 2018-12-28:
Lost track of this issue, so I am a bit late to confirm. But yes, all good now :)