dotintent/react-native-ble-plx #367: [Android] - Fatal Exception: rx.exceptions.OnErrorNotImplementedException

GroupXTech, 2018-11-21:
I seeing this issue randomly.
The specific devices were ONEPLUS A6003, Pixel, Pixel 2. 
Attached are the crash logs from Fabric.

Im also getting several complaints that multiple android devices are having long connect times. 
I've implemented automatic reconnects if it fails either in the connect or service discovery. 
worst case connect times have been reported up to 10 mins. I've only ever seen up to 1 min. But that is even a bit too long... I do see gatt error 133 a lot. 

Any insight is greatly appreciated.

Regards.

[com.teeter_issue_50_crash_5BF54B110283000143D7624CA2EA0323_DNE_0_v2.txt](https://github.com/Polidea/react-native-ble-plx/files/2604381/com.teeter_issue_50_crash_5BF54B110283000143D7624CA2EA0323_DNE_0_v2.txt)

Cierpliwy, 2018-11-23:
@dariuszseweryn Can you look at logs? It seems related to Rx issues inside RxAndroidBle, but I'm not sure. Issue is pointing at following code:
```
        /*
         The below .subscribe() is only to make the above .cache() to start working as soon as possible.
         We are not tracking the resulting `Subscription`. This is because of the contract of this class which is supposed to be called
         when a disconnection happens from one of three places:
            1. adapterStateObservable: the adapter turning into state other than STATE_ON
            2. onDisconnectedException
            3. onGattConnectionStateException
         One of those events must happen eventually. Then the adapterStateObservable (which uses BroadcastReceiver on a Context) will
         get unsubscribed. The rest of this chain lives only in the @ConnectionScope context and will get Garbage Collected eventually.
         */
        disconnectionErrorOutputObservable.subscribe();
    }
```
Probably `disconnectionErrorOutputObservable` emitted error here.

GroupXTech, 2018-12-02:
Any update for this?
Could this also be causing an excessive time to connect?

dariuszseweryn, 2019-01-02:
Hello, sorry for such a delay.

> Could this also be causing an excessive time to connect?

Unlikely
> Probably `disconnectionErrorOutputObservable` emitted error here.

I analysed the code and there must not be any errors being emitted from this `disconnectionErrorOutputObservable` under normal circumstances. The issue here seems to be this:
> Caused by java.lang.IllegalStateException: Too many receivers, total of 1000, registered for pid: 17367, callerPackage: com.teeter

As far as I know the library does not leak broadcast receivers (these are used only in `RxBleAdapterStateObservable` and `LocationServicesOkObservableApi23` but are unregistered there as well) so it may be a problem that started in a different place but only manifested itself while using the library

closed 2019-03-23 by stale[bot]

mfpiccolo, 2019-09-20:
We are experiencing this issue with 1.0.3

dariuszseweryn, 2019-09-21:
@mfpiccolo Could you add any actionable information? Under what circumstances this happens? Logs with the library logging enabled?

reinhardholl, 2020-09-03:
We are also seeing this issue on `1.1.1`. For some more context @dariuszseweryn we have implemented an auto retry when failing to connect to a device. I noticed the retries sometimes run away (which is an issue on our side), but after a few attempts we get: 
```
rx.exceptions.OnErrorNotImplementedException
Too many receivers, total of 1000, registered for pid: 28916, callerPackage
```

PaulB2Code, 2020-09-30:
We also facing this issuer on : "react-native-ble-plx": "^2.0.1"

  const startDeviceScan = () => {

    console.trace('startDeviceScan')
    let i = 0;
    bleManager.startDeviceScan(null, { allowDuplicates: false }, (error, device) => {
      if (error) {
        console.error(error)
        seterrorInfo(error)
      } else {
        seterrorInfo("ADD Device" + i)
        i = i + 1
        if (i > 10) {
          bleManager.stopDeviceScan()
        }
        addDevice(device)
      }
    })
  }

It happens when we are scanning. Not all the time. 

We use a pixel 2 - Android 11.

Ilphrin, 2021-01-29:
Hi!

We are facing the same issue here on multiple devices, (but seems to be working well on a OnePlus 6T), all with Android 11. Our app is working for the first 10s and then crashes with the exact same error everytime.

Is there any news about this issue ? Or perhaps can we provide some help to figure out somehow what is going on?

(Thanks you btw for this cool library!)

yeapkrush, 2021-02-07:
Hi I get the same issue, I'm also using OnePlus 6T. @Ilphrin did you manage to find the reason? Much appreciated...

Ilphrin, 2021-02-16:
@yeapkrush Nope, we changed to another bluetooth library sadly, as we had to deliver our POC as soon as possible :/

yeapkrush, 2021-02-18:
**For anyone who reads this in the future:**
**For me the problem was:** I was **setting the state** inside the listener function. My BLE device usually sends data frequently so after around 1000 times the listener function gets called, I get the "Too many receivers, total of 1000, registered for pid: XXXXX" error.
I don't know why it happens but the way **I solved it is by using useRef instead of useState inside the listener function**.

asaf-demetria, 2021-02-21:
Hi,
Happens to me as well. Here's the full stack trach:
```
rx.exceptions.OnErrorNotImplementedException: Too many receivers, total of 1000, registered for pid: 7243, callerPackage: ag.demetria.app
        at rx.internal.util.InternalObservableUtils$ErrorNotImplementedAction.call(InternalObservableUtils.java:386)
        at rx.internal.util.InternalObservableUtils$ErrorNotImplementedAction.call(InternalObservableUtils.java:383)
        at rx.internal.util.ActionSubscriber.onError(ActionSubscriber.java:44)
        at rx.observers.SafeSubscriber._onError(SafeSubscriber.java:153)
        at rx.observers.SafeSubscriber.onError(SafeSubscriber.java:115)
        at rx.internal.operators.OnSubscribeMap$MapSubscriber.onError(OnSubscribeMap.java:88)
        at rx.Observable.unsafeSubscribe(Observable.java:10334)
        at rx.internal.operators.OnSubscribeMap.call(OnSubscribeMap.java:48)
        at rx.internal.operators.OnSubscribeMap.call(OnSubscribeMap.java:33)
        at rx.Observable.subscribe(Observable.java:10423)
        at rx.Observable.subscribe(Observable.java:10390)
        at rx.Observable.subscribe(Observable.java:10195)
        at com.polidea.multiplatformbleadapter.BleModule.monitorAdapterStateChanges(BleModule.java:1120)
        at com.polidea.multiplatformbleadapter.BleModule.createClient(BleModule.java:114)
        at com.polidea.reactnativeble.BleClientManager.createClient(BleClientManager.java:83)
        at java.lang.reflect.Method.invoke(Method.java:-2)
        at com.facebook.react.bridge.JavaMethodWrapper.invoke(JavaMethodWrapper.java:372)
        at com.facebook.react.bridge.JavaModuleWrapper.invoke(JavaModuleWrapper.java:151)
        at com.facebook.react.bridge.queue.NativeRunnable.run(NativeRunnable.java:-2)
        at android.os.Handler.handleCallback(Handler.java:883)
        at android.os.Handler.dispatchMessage(Handler.java:100)
        at com.facebook.react.bridge.queue.MessageQueueThreadHandler.dispatchMessage(MessageQueueThreadHandler.java:27)
        at android.os.Looper.loop(Looper.java:214)
        at com.facebook.react.bridge.queue.MessageQueueThreadImpl$4.run(MessageQueueThreadImpl.java:226)
        at java.lang.Thread.run(Thread.java:919)

Caused by: java.lang.IllegalStateException: Too many receivers, total of 1000, registered for pid: 7243, callerPackage: ag.demetria.app
        at android.os.Parcel.createException(Parcel.java:2079)
        at android.os.Parcel.readException(Parcel.java:2039)
        at android.os.Parcel.readException(Parcel.java:1987)
        at android.app.IActivityManager$Stub$Proxy.registerReceiver(IActivityManager.java:4826)
        at android.app.ContextImpl.registerReceiverInternal(ContextImpl.java:1527)
        at android.app.ContextImpl.registerReceiver(ContextImpl.java:1488)
        at android.app.ContextImpl.registerReceiver(ContextImpl.java:1476)
        at android.content.ContextWrapper.registerReceiver(ContextWrapper.java:627)
        at android.content.ContextWrapper.registerReceiver(ContextWrapper.java:627)
        at com.polidea.rxandroidble.RxBleAdapterStateObservable$1.call(RxBleAdapterStateObservable.java:61)
        at com.polidea.rxandroidble.RxBleAdapterStateObservable$1.call(RxBleAdapterStateObservable.java:47)
        at rx.internal.operators.OnSubscribeCreate.call(OnSubscribeCreate.java:72)
        at rx.internal.operators.OnSubscribeCreate.call(OnSubscribeCreate.java:32)
        at rx.Observable.unsafeSubscribe(Observable.java:10327)
        at rx.internal.operators.OnSubscribeMap.call(OnSubscribeMap.java:48)
        at rx.internal.operators.OnSubscribeMap.call(OnSubscribeMap.java:33)
        at rx.Observable.subscribe(Observable.java:10423)
        at rx.Observable.subscribe(Observable.java:10390)
        at rx.Observable.subscribe(Observable.java:10195)
        at com.polidea.multiplatformbleadapter.BleModule.monitorAdapterStateChanges(BleModule.java:1120)
        at com.polidea.multiplatformbleadapter.BleModule.createClient(BleModule.java:114)
        at com.polidea.reactnativeble.BleClientManager.createClient(BleClientManager.java:83)
        at java.lang.reflect.Method.invoke(Method.java:-2)
        at com.facebook.react.bridge.JavaMethodWrapper.invoke(JavaMethodWrapper.java:372)
        at com.facebook.react.bridge.JavaModuleWrapper.invoke(JavaModuleWrapper.java:151)
        at com.facebook.react.bridge.queue.NativeRunnable.run(NativeRunnable.java:-2)
        at android.os.Handler.handleCallback(Handler.java:883)
        at android.os.Handler.dispatchMessage(Handler.java:100)
        at com.facebook.react.bridge.queue.MessageQueueThreadHandler.dispatchMessage(MessageQueueThreadHandler.java:27)
        at android.os.Looper.loop(Looper.java:214)
        at com.facebook.react.bridge.queue.MessageQueueThreadImpl$4.run(MessageQueueThreadImpl.java:226)
        at java.lang.Thread.run(Thread.java:919)
```

Seems like it happens after a callback (in my case `startDeviceScan`) has been callback multiple times (the use case is when there is one BT found and the callback is being called over and over with the same `Device`.

reinhardholl, 2021-05-28:
Any update on this one? We are still seeing this issue after a year :(

yeapkrush, 2021-05-28:
Hey!
Umm, I finally realized that this was due to poor coding on my side and not from react-native-ble-plx.
The way I had initialized ble was wrong. 
When I was using this package for the first time I put `const manager = new BleManager();` in the component itself so every time the component re-rendered a new ble manager was created. I noticed this after few months. (maybe you have done a better job than I did :) ). A very careless mistake.
This is the main issue for the problem because this will cause a huge memory leak. The solution that I posted before is a very very bad fix for this (I wouldn't even call it a fix). 
Try to move `const manager = new BleManager();` to the top of the code after you import `react-native-ble-plx`. If it works then you have the same issue I had. So if you do this make sure that you pass the manager to child components rather than initializing again in child components.

The better way to initialize is by creating a class for ble.
See here: [https://medium.com/@polidea/how-to-code-bluetooth-low-energy-device-with-react-native-fbcf76c7e821](url)
But what I did was; I created an instance of the ble manager as a state in the redux store so that I can access it from any component. I found this method much easier to manage as well.

danish-danial, 2024-07-18:
Hello, I'm not sure this will help. But for my case, it turn out the `manager` didn't stop scanning the devices.

```JS
  const scanDevices = async () => {
    const isPermissionsEnabled = await requestPermissions();
    const bluetoothState = await BluetoothStateManager.getState();

    if (isPermissionsEnabled && bluetoothState.toString() === "PoweredOn") {
      setIsScanning(true);
      manager.startDeviceScan(
        null,
        { scanMode: ScanMode.LowLatency },
        (error, device) => {
          if (error) {
            throw error;
          }

          if (device?.name?.includes("GoPro")) {
            setAllDevices((prevState: Device[]) => {
              if (!isDuplicateDevice(prevState, device)) {
                return [...prevState, device];
              }
              setIsScanning(false);
              manager.stopDeviceScan();
              return prevState;
            });
          }
        },
      );
    }

```