dotintent/react-native-ble-plx #368: how to set writeCharacteristicWithResponseForDevice() argument?

whhotw, 2018-11-21:
I didn't way is right that write hex string to my board, anybody know that? i did a test my board by free BLE scanner, it's work.
Please refer screenshot that write '1b01' hex string. how can i do same thing on my app?
[image]

it's a BLE scanner _NOT_ my app, just for test
[image]

Cierpliwy, 2018-11-23:
Values are accepted in Base64 format, so you need to convert your hexadecimal string representation to Base64 in:

- binary form: `1B02` is equal to `GwI=`. 
- string form: `1B02` -> `31423032` -> `MUIwMg==` / `MUIwMgA=` (null terminated)

whhotw, 2018-11-23:
@Cierpliwy it's work, thank you so much!

closed 2018-11-23 by whhotw