dotintent/react-native-ble-plx #369: Unable to find services for my device

ssajnani, 2018-11-21:
I keep getting null for my serviceUUIDs:
``
`id: "B8:27:EB:F8:F2:67"
isConnectable: null
localName: null
manufacturerData: null
mtu: 23
name: "lifepi0000"
overflowServiceUUIDs: null
rssi: null
serviceData: null
serviceUUIDs: null
solicitedServiceUUIDs: null
txPowerLevel: null
```

Also no services found:
```
BleError: Services not discovered for device ..
```

When I run the same check using another package react-native-ble-manager, everything seems to work fine and it finds my services. Does anyone know what wrong? Is there a specific version of react-native etc that I need to run. Thanks!

ssajnani, 2018-11-23:
The first output above is from running discoverAllServicesAndCharacteristicsForDevice. I'm getting that data from the promise.

Cierpliwy, 2018-11-23:
Can you give me a code snippet and native logs?

ssajnani, 2018-11-23:
this.manager.connectToDevice(deviceID) 
  .then((device) => { 
    device.discoverAllServicesAndCharacteristics()
      .then((results) => { 
        console.log(results);
        this.manager.servicesForDevice(deviceID)
          .then((services) => {
            console.log(services);
          });
      });
});

hidayatNgganteng, 2020-02-20:
i am change 

> device.discoverAllServicesAndCharacteristicsForDevice()

to 

> this.manager.discoverAllServicesAndCharacteristicsForDevice(peripheralId)

------------------------------------
and  change

> device.servicesForDevice()

to 

> this.manager.servicesForDevice(peripheralId)

------------------------------------

**and the problem is resolved**

renatoramossilva, 2020-04-05:
> i am change
> 
> > device.discoverAllServicesAndCharacteristicsForDevice()
> 
> to
> 
> > this.manager.discoverAllServicesAndCharacteristicsForDevice(peripheralId)
> 
> and change
> 
> > device.servicesForDevice()
> 
> to
> 
> > this.manager.servicesForDevice(peripheralId)
> 
> **and the problem is resolved**

Hi @hidayatNgganteng !

I have the same problem.. I can scan and connect the device, but when I try to list the services I receve the same message error.

Can you show in your code how you solved this issue?

Thansks

hitanshuGajjar20, 2021-04-20:
**device.discoverAllServicesAndCharacteristicsForDevice()**
for this thing i am getting error of Function not found 

**this.manager.discoverAllServicesAndCharacteristicsForDevice(peripheralId)**
and in this i am getting the same response as connectToDevice ( null values )