dotintent/react-native-ble-plx #372: Why startDeviceScan(['180F'], null, (error, device) => (...)) works on iOS but not on Android?

lorenc-tomasz, 2018-11-28:
Hi,
i try to search devices using service UUIDs. It works very well on iOS, but not on android devices. Why?

leo-tavares, 2018-11-29:
I use the scanner on android and it works perfectly.
Could you describe the problem in more detail?

lorenc-tomasz, 2018-11-30:
I get the example https://github.com/Cierpliwy/SensorTag and tried to search devices by service UUID, in this case battery one. On iOS it works great, but on Android each device looks like this:
id: "xx:xx:xx:xx:xx"
isConnectable: null
localName: null
manufacturerData: "TAAQBQMcNQz3"
mtu: 23
name: "BLE Battery tester"
overflowServiceUUIDs: null
rssi: -51
serviceData: null
serviceUUIDs: null
solicitedServiceUUIDs: null
txPowerLevel: null

Cierpliwy, 2018-11-30:
Please set breakpoint at BleModule.java:416 (ver 1.0.1), that's the line with following content:
`String deviceId = scanResult.getBleDevice().getMacAddress();`
Then send me content of `scanResult.getScanRecord().getBytes()`. I want to make sure that your serviceUUIDs are parsed properly.

lorenc-tomasz, 2018-12-03:
OK. Will try do this tomorrow.

closed 2019-01-04 by dariuszseweryn