dotintent/react-native-ble-plx #373: Multiple bluetooth devices with same MAC address

kenobi91, 2018-11-29:
I have a situation where I need to connect two devices to my app but both of them have a same mac address. I have tried AT commands, but I only can change the name of the device. Is there a way to connect to the device based on id (mac address) and, for instance a name or some other way to change the address of the device?

Cierpliwy, 2018-11-30:
Every Bluetooth device should have unique address to operate properly. If you have public address then your manufacturer should use their own pool of addresses and associate them with specific devices.

Otherwise you can use random static addresses:
"A device may choose to initialize its static address to a new value after each power cycle. A device shall not change its static address value once initialized until the device is power cycled."

You can read more about it in Bluetooth Core Specification Vol 6, part B, 1.3.

kenobi91, 2018-11-30:
Thank you for answer i will look this up :)

closed 2019-01-04 by dariuszseweryn