dotintent/react-native-ble-plx #375: Connection latency on iOS

baransu, 2018-12-06:
I heavy really on `monitorCharacteristicForService` with the low latency.

After the connection and discovering characteristics, I request High connection priority. It works fine on Android and the latency is really small but on iOS I experience high latency. I cannot tell how much exactly but it's around 1 second. There was the same problem with Android but requesting connection priority fixed the issue. Looking at source code `requestConnectionPriority` on iOS does nothing. If it's possible to implement it somehow I would really like to contribute 🙂

baransu, 2018-12-06:
If anyone is experiencing similar issues I improved the performance by not using `monitorCharacteristicForService` and creating polling implementation using `readCharacteristicForService`.

dariuszseweryn, 2019-01-04:
> If it's possible to implement it somehow I would really like to contribute

Just found out that [there seems to be a way](https://developer.apple.com/documentation/corebluetooth/cbperipheralmanager/1393277-setdesiredconnectionlatency)

closed 2019-03-23 by stale[bot]

julioxavierr, 2019-08-12:
Any updates on this?

dariuszseweryn, 2019-08-16:
Not possible to do this on iOS as far as I know. The above link is for iOS as a peripheral device and this library is for central device use case.