dotintent/react-native-ble-plx #380: isDeviceConnected and device.isConnected not returning True/False value

jkpvecino, 2018-12-17:
I'm currently trying to see if the package can detect if a device that's currently connected through the OS's Bluetooth module is also seen as connected within the app. When I call isDeviceConnected using the correct MAC address or calling device.isConnected after scanning for the device I'm looking for, I get the following value instead of a boolean:

{"_40":0,"_65":0,"_55":NULL,"_72":NULL}

This looks like a weirdly-formatted MAC address but I'm not totally sure since it doesn't look like the MAC address of the device.

jclrosa, 2018-12-27:
On my case, I connected to a BLE device through the IOS Bluetooth menu (outside the app context), then when I enter the app, I need to know which is the device I am connected to. 

When I do what I described above, the connected device does not show on the scanAndConnect method, which is expectable, since the device is not `available` anymore. However I would like to know what is the device that I am connected to.

I tried `connectedDevices()`, without passing any device_id on the arguments array, because I don’t have any device_id to compare to, but the answer is always empty.  I just need to know what is the device that I am connected if this is the case. Is there any way to achieve it?

dariuszseweryn, 2019-01-04:
Currently the library only tracks what it connected to. There is no way to find what is connected to the mobile.

closed 2019-03-23 by stale[bot]

fitbuff, 2020-08-18:
@jclrosa have you found a workaround for this?